XKNX/xknx #1: Home Assistant

twistermcneal, 2016-11-21:
Is it possible to implement this in Home Assistant ?

I'am trying to use a Shutter/Cover component for the KNX Bus...
does this one work?

Thanks in advance

Julius2342, 2016-11-21:
Yes, thats the plan. I have already a home assistant plugin to control the lights, shutter will follow.

twistermcneal, 2016-11-21:
Thanks for replay!

You use a different repo for the Plugin ?

Julius2342, 2016-11-22:
I added what i use at the moment to bring some light to my construction site. This is just a stub as a basis for discussion :-)

Julius2342, 2016-12-06:
will publish an updated version of my plugin very soon.

Julius2342, 2016-12-08:
May you try my new version? The plugin does not have shutter support yet. This is next on my roadmap.

twistermcneal, 2016-12-08:
Yes sure i try it on weekend ..and give you some Feedback. :)

Julius2342, 2016-12-09:
i added some basic shutter support ...

twistermcneal, 2016-12-10:
i have tested he xknx lib a little bit... it works i can switch lights on/off. the config part is a little bit confusing when you only use homeassistant .. i think we should change this to config each device in the home assitant style:

configuration.yaml
`
xknx:
    own_address: "15.15.249"
    own_ip: "192.168.2.164"
`
`
switch:       
        Livingroom.Switch_1:
            group_address: 1/2/0
            group_address_state: 1/2/1`

twistermcneal, 2016-12-10:
i have also made some changes to get it run 

devices.py
> def device_by_group_address( self, group_address):
        for device in self.devices:
            if device.has_group_address(group_address):
                return device
        return None

multicast.py:

>  device = devices_.device_by_group_address(telegram.group_address)
                    #Ignoring unkown devices (can be used as discover funktion)
                    if device != None:                      
                        device.process(telegram)
                        if ( callback ):
                            callback(device,telegram)

thats because some devices are unkown

twistermcneal, 2016-12-10:
i also recommend to use a value cache system thats because some devices are only sending the state on the bus but you can not read it !

i try to use git hub to do some changes and make a pull request but its my first time using github!

Julius2342, 2016-12-10:
I would personally prefer to keep the exception, bc then you can do things like:

 ```
    try:
       if (device.name == "Livingroom.Switch_2" ):
            if device.is_on():
                devices_.device_by_name("Livingroom.Shutter_1").set_short_down()
                devices_.device_by_name("Livingroom.Outlet_2").set_on()
            elif device.is_off():
                devices_.device_by_name("Livingroom.Shutter_1").set_short_up()
                devices_.device_by_name("Livingroom.Outlet_2").set_off()
    except CouldNotResolveAddress as c:
        print(c)
```

without handling the `None` return value in each line. Would you agree to add an except mechanism within `multicast.py` ?

twistermcneal, 2016-12-10:
yes of course i think its the better way.
I tryed to commit my changes ro the dev branch, but its my first time using github so i dont understand everything hope i dont messed up something

Julius2342, 2016-12-10:
great, i will merge to master and dev!

No you did not mess up anything :-)

Julius2342, 2016-12-10:
regarding the configuration, im a bit undecided, bc my xknx.yml will get very, very large (several hundred lines :-D ) and i want to keep the functionality to start it with separate python scripts. On the other hand I see your point.

What kind of devices do you control via KNX?

twistermcneal, 2016-12-10:
i control shutters, Lights(Dimmer,RGB), Climate and Weather, Alarm  and some media devices via my knx system. Also all Sensors like Doors and  Windows.

I think the configuration is not the problem, only the first step is a little bit confusing.

Julius2342, 2016-12-10:
Dimmers will come next. :-)

twistermcneal, 2016-12-10:
Nice :)

twistermcneal, 2016-12-10:
i try to do some changes too, but im not a python programmer so i think i can only give you some suggestions

Julius2342, 2016-12-10:
highly appreciated! I think you have to fork the repo and then you can push your changes to your own remote branches. (After you have forked i think you have to change your origin)

Julius2342, 2016-12-10:
Btw do you understand the difference between short time operation and stop in KNX protocol? Btw how do you trigger a stop if you also want to have short time operations ...

twistermcneal, 2016-12-10:
There is no diffrence its used as stop and move...
If the shutter is in state running it will trigger the stop after that you can start the tilt or move with the short....short(1) moves the shutter only a specified amount up if you want to stop the short opertation you have to trigger the short(0)

twistermcneal, 2016-12-10:
how can  i implement a callback in pyhon that triggers the update_ha_state so the bus event is triggered immediately? 
in your device i found something like  self.after_update_callback = lambda x: None

Julius2342, 2016-12-11:
im adding this for cover.py at the moment. The mechanism works like this:

The line you cited

```
self.after_update_callback = lambda x: None
````

defines the callback function and sets it to an empty function, doing nothing.

When the xknx server receives an update, it calls this function.:

```
        self.after_update_callback(self)
````

In the outside logic (the ha-plugin) you can override this function with any function you want. I use it to trigger the bus event:

```
    def __init__(self, hass, device):
[...]
        self.register_callbacks()

    def register_callbacks(self):
        def after_update_callback(device):
            self.update()
        self.device.after_update_callback = after_update_callback

    def update(self):
 [...]
        self.update_ha_state()

twistermcneal, 2016-12-11:
thanks for explanation i think i got it...

Julius2342, 2016-12-11:
now we have a conflict :-/

twistermcneal, 2016-12-11:
oh you already did some changes o the cover didnt see this

Julius2342, 2016-12-11:
I just committed my changes ...

Julius2342, 2016-12-11:
May you retry?

closed 2016-12-14 by Julius2342