XKNX/xknx #5: Shutters with our position-group address

Julius2342, 2016-12-14:
There are shutters which don't support an absolute position group address.

I think we should do somehting like:

````
    class Direction(Enum):
        UP = 1
        DOWN = 2

    def set_position(self, position):
        if self.group_address_position is not None:
            self.send(self.group_address_position, [0x80, position])
        else
            self.set_position_relative(position)

    dev set_position_relative(self, position):
        relative_position = position - self.position
        direction = Direction.UP if relative_position > 0 else Direction.DOWN

        travel_time_full = travel_time_up if direction == Direction.UP else travel_time_up_down # should be taken from configuration ..

        travel_time = travel_time_full * abs(relative_position) / 100
        self.travel( direction, travel_time)
 
    dev travel(self, direction, travel_time ):

        if direction == Direction.UP:
           self.set_up()
        else:
           self.set_down()

       time.sleep(travel_time)

       self.set_short_down()
````

The problem with that approach is, that if changed in between, you cant readjust the cover. We therefore should lock the device.

twistermcneal, 2016-12-14:
i think we should do this like in the demo cover (  _listen_cover )
i dont know how exactly track_utc_time_change runs but in my first approach this works very well i can start the movement and stop it between, also from bus side. Or we use a timestamp and calculate the current pos in a own thread, while it moves

Julius2342, 2016-12-14:
Ahh good idea! I will have a look in demo cover again!

Julius2342, 2016-12-17:
May you test? branch: cover-without-absolute-position-t5

you have to change your xknx.yaml:

```
    shutter:
        # Shutters without direct positioning:
        Livingroom.Shutter_3: {group_address_long: "1/4/9", group_address_short: "1/4/10", group_address_position_feedback: "1/4/11", travelling_time_down: 50, travelling_time_up: 60 }
```

twistermcneal, 2016-12-25:
Nachdem ich die pyc Dateien gelöscht habe sind momentan keine Fehler vorhanden!
Trigger von Bus Seite muss noch implementiert werden!

closed 2017-01-01 by Julius2342