XKNX/xknx #6: Central dimmers should not have any "is_open" or "is_closed" state

Julius2342, 2016-12-14:
For central dimmers it is not possible to determine a general is_open or is_closed state. Therefore both values have to return a None.

To make this possible, central dimmers have to be marked as such in the configuration.

twistermcneal, 2016-12-14:
if we own all devices, i think we can loop thru all devices, and get the current position of a defined group
something in the config like groupId = A for each device 
the central shutter use this id to get the right states
just my two cents

Julius2342, 2018-10-18:
A very old issue :)

I think the KNX concept of having a central switch conflicts with the HASS approach.

I solved this in my setup by not using a central KNX address but instead having a KNX-binary input and then switching all lights on/off all covers up/down.

This also has the advantage, that all states (of each light/cover) is instantly available within hass.

spali, 2018-10-18:
Yes, I know it's really old, but wonder why not more people have this "problem".
It's one of the base model how to solve central functions in KNX except scenes.

I'm thinking about to solve it by scenes, but I hassle with that, because it costs scene numbers in KNX and makes it a lot harder to maintain in KNX.

Still think something generic like the "stateless" option could be a real improvements, but don't know if HA can support this out of the box.

But I'm curious about how you solved it... what do you mean with "KNX-binary input"?
Take your time with the answer, I don't want to disturb you on your traveling ;)

Julius2342, 2018-10-19:
This example turns on/off the hue_strips when pressed once and all lights in the house if pressed twice: 

```
- platform: knx
  name: Sleeping.SwitchLeft
  address: '5/0/6'
  automation:
    - counter: 1
      hook: 'on'
      action:
       - entity_id: light.hue_lightstrip_1
         service: homeassistant.turn_on
       - entity_id: light.hue_lightstrip_2
         service: homeassistant.turn_on
       - entity_id: light.hue_lightstrip_3
         service: homeassistant.turn_on
    - counter: 2
      hook: 'on'
      action:
       - service: light.turn_on

- platform: knx
  name: Sleeping.SwitchRight
  address: '5/0/7'
  automation:
    - counter: 1
      hook: 'on'
      action:
        - entity_id: light.hue_lightstrip_1
          service: homeassistant.turn_off
        - entity_id: light.hue_lightstrip_2
          service: homeassistant.turn_off
        - entity_id: light.hue_lightstrip_3
          service: homeassistant.turn_off
    - counter: 2
      hook: 'on'
      action:
        - service: light.turn_off
```

spali, 2018-11-01:
Hi @Julius2342 
sorry for the late response, was travelling too ;)
In my case, I need it the other way around.
I wan't one a "central" cover in the interface. And with your suggestion I would implement a switch just in HA and then have actions for all covers or lights with the specific group addresses.

take this use-case:
If I want a "Go to bed" button which turns off all lights (around 40 in the whole house) and closes all covers (16) .... it would flood the KNX bus with 56 telegrams initiated by HA.

That is exactly what you want to prevent with KNX central group addresses which has no status and only requires one KNX telegram.

farmio, 2021-01-25:
This can be achieved via scenes or in HA with customizing a cover with `assumed_state: True`.

closed 2021-01-25 by farmio