XKNX/xknx #10: DPT Types

twistermcneal, 2016-12-16:
I use this as DPT reference. Should we use this for documentation purposes?

|          Function          |  length |     EIS    |     DPT    |             Value            |
|:----------------------------------:|:-------------------:|:----------:|:----------:|:----------------------------:|
| Switch                             | 1 Bit               | EIS 1      | DPT 1      | 0,1                          |
| Dimming (Position, Control, Value) | 1 Bit, 4 Bit, 8 Bit | EIS 2      | DPT 3      | [0,0]...[1,7]                |
| Time                               | 3 Byte              | EIS 3      | DPT 10     |                              |
| Date                               | 3 Byte              | EIS 4      | DPT 11     |                              |
| Floating point                     | 2 Byte              | EIS 5      | DPT 9      | -671088,64 - 670760,96       |
| 8-bit unsigned value               | 1 Byte              | EIS 6      | DPT 5      | 0...255                      |
| 8-bit unsigned value               | 1 Byte              | DPT 5.001  | DPT 5.001  | 0...100                      |
| Blinds / Roller shutter            | 1 Bit               | EIS 7      | DPT 1      | 0,1                          |
| Priority                           | 1 Bit               | EIS 8      | DPT 2      | [0,0]...[1,1]                |
| IEEE Floating point                | 4 Byte              | EIS 9      | DPT 14     | 4-Octet Float Value  |
| 16-bit unsigned value              | 2 Byte              | EIS 10     | DPT 7      | 0...65535                    |
| 16-bit signed value                | 2 Byte              | DPT 8      | DPT 8      | -32768...32767               |
| 32-bit unsigned value              | 4 Byte              | EIS 11     | DPT 12     | 0...4294967295               |
| 32-bit signed value                | 4 Byte              | DPT 13     | DPT 13     | -2147483648...2147483647     |
| Access control                     | 1 Byte              | EIS 12     | DPT 15     |                              |
| ASCII character                    | 1 Byte              | EIS 13     | DPT 4      |                              |
| 8859_1 character                   | 1 Byte              | DPT 4.002  | DPT 4.002  |                              |
| 8-bit signed value                 | 1 Byte              | EIS 14     | DPT 6      | -128...127                   |
| 14 character ASCII                 | 14 Byte             | EIS 15     | DPT 16     |                              |
| 14 character 8859_1                | 14 Byte             | DPT 16.001 | DPT 16.001 |                              |
| Scene                              | 1 Byte              | DPT 17     | DPT 17     | 0...63                       |
| HVAC                               | 1 Byte              | DPT 20     | DPT 20     | 0..255                       |
| Unlimited string 8859_1            | .                   | DPT 24     | DPT 24     |                              |
| List 3-byte value                  | 3 Byte              | DPT 232    | DPT 232    | RGB[0,0,0]...[255,255,255]   |

Julius2342, 2016-12-16:
Stupid question: What does DPT and EIS stand for?

twistermcneal, 2016-12-16:
Data Point Types / EIB Interworking Standard

twistermcneal, 2016-12-16:
[Datapoint.pdf](https://github.com/Julius2342/xknx/files/658217/Datapoint.pdf)

Julius2342, 2016-12-16:
danke!

twistermcneal, 2016-12-16:
parametrierst du deine Geräte selber mit ETS oder erledigt das eine Firma?

Julius2342, 2016-12-16:
What I don't understand:

I wanted to crosscheck if I implemented the temperature correctly. I get three bytes as payload:

80 05 94

80 is the status field and the temperature is encoded in 05 94. Which is at the moment: (5*256+148)/100= 14.28.

If I calculate correctly the minimum value of this approach is 0.00 C and the maximum value should be (255*256+256/)100=655.36.

Page 117 of your document claims to encode -273 C - 655,34 C ... How is this possible? Or do they use V16 to encode some magic?

Julius2342, 2016-12-16:
ich parametriere selbst :-)

twistermcneal, 2016-12-16:
Thats an 16Bit signed integer value
the range is defined as –273°C to 655,34

bit0 - bit15 = value
bit16 = if true +, if false -

if you have the value you have to calculate the real value according to the max and min defined in the doc!

somthing like: 
norm = ((value - minInteger) / (maxInteger - minInteger))

temp = ((norm / (maxDPT - minDPT)) + minDPT

closed 2017-01-01 by Julius2342