XKNX/xknx #11: Send Time to KNX bus

Julius2342, 2016-12-16:
We have to send the time to the KNX bus to display it properly within my [GIRA Tastsensor 3 Plus 2-fach 514200 ](https://katalog.gira.de/de_DE/datenblatt.html?id=635019)

Open question:

- Is time a device? 
- Do we send UTC or do they use local time
- How should the configuration look like

````
    time:

        Time_1:
            group_address: "1/2/7"
            time_zone: "Europe/Berlin"  # If the bus needs local time
            frequency: 20 # send every 20 seconds
`````
 
Alternatively we could put the timing to the hass_component (using this utc_changed ... function)

twistermcneal, 2016-12-16:
- My weather station sends the time and date, general i think its a device!
- I think we should make this configureable . Some devices need UTC others local time.
- Config
` if time_zone is None use UTC`

Julius2342, 2016-12-28:
Closing this, at the moment sending only localtime, will change this if anybody needs UTC ...

closed 2016-12-28 by Julius2342