XKNX/xknx #8: Adding a proper input/output queue

Julius2342, 2016-12-16:
In order to make unit testing we need a proper input/output queue to be filled and read from unit tests.

This would also enable us to implement proper locking on the event system.

````
global Queue
````

within devices we do:

```
with Queue.lock:
     Queue.out( telegram )
     Queue.process_out()
```

within Multicast receiver we do:

`````
with Queue.lock:
     Queue.in( telegram )
     Queue.process_in()
````

Alternatively we could have a separate thread for handling both process input queue and output queue.

Within unit tests we could e.g. do device.set_on() and then test the correct telegram in the queue.

twistermcneal, 2016-12-16:
Do you want to write your own queue class 
https://docs.python.org/2/library/queue.html
sounds stright forward

Julius2342, 2016-12-16:
Oh nice :) will have a look!

(im pretty new to python, so i have to get familiar with the standard approaches...)

twistermcneal, 2016-12-16:
Wir können ja auch mal deutsch reden .... mein englisch ist eh nicht mehr das beste :)

Aber zur Übung  kanns auch  nicht schaden...
Grüße aus Bayern
Michael

Julius2342, 2016-12-16:
haha :) bin auch aus Bayern.

Julius2342, 2016-12-16:
seit wann bastelst Du mit KNX?

twistermcneal, 2016-12-16:
Seit gut 15 Jahren immer wieder mal, seit 7 Jahren hab ichs bei mir im Haus und erweitere immer weiter!
Momentan bin ich eigentlich gerade dabei eine Lösung für mein Raumfeld System(Multiroom) zu entwickeln.
Da ich Home-Assistant für mich entdeckt habe, verwende ich hierfür auch Python.

Naja immer wieder was neues dafür hat man ja ein Bus System !

Julius2342, 2016-12-16:
ich erst seit kurzem - bin noch nicht mal eingezogen. Ich hoffe auf Januar :)

twistermcneal, 2016-12-16:
Dann kann man ja schon mal zum Eigenheim gratulieren !
Und denk immer daran „Das Haus ist eine Maschine zum Wohnen. Ein Sessel ist eine Maschine zum Sitzen.“

:)

closed 2016-12-28 by Julius2342