XKNX/xknx #9: Implement target temperature in termostat module

Julius2342, 2016-12-16:
The target temperature is not yet submitted to the KNX bus.

I also have to find out if you get a KNX event if the target temperature is changed within the Thermostat itself.

Additionally we have to read the normal update Telegrams and update the local target temperature.

twistermcneal, 2016-12-16:
In my case i have 3 group addresses (6326-101 trition). 

> scenario 1

**base_setpoint** is used to set the new temp from a central unit
**actual_temperature** sends acutal temp on bus if changed, **can not be read**
**actual_setpoint** sends actual setpoint to bus, if changed, **can not be read**

But in general we have the problem that this can be changed in the ets config.

> scenario 2

**base_setpoint** is used to set the new temp from a central unit
**actual_temperature** sends acutal temp on bus if changed, **can be read**
**_no_** **actual_setpoint** 

> scenario 3

**base_setpoint** is used to set the new temp from a central unit
**_no_** **actual_temperature** 
**actual_setpoint**  sends actual setpoint to bus, if changed, **can be read**

i think the most common scenario is 1 for this thermostat

Julius2342, 2017-10-05:
@twistermcneal : Would you have time to chat about these use cases in the next couple of days?

twistermcneal, 2017-10-05:
yes of course i think we can discuss this on the weekend

Julius2342, 2017-10-05:
That would be great!

Julius2342, 2017-10-07:
Hi @twistermcneal : when would you have time?

twistermcneal, 2017-10-09:
sry did not make it at the weekend... iam in the gitter chat at the moment if you have time?

Julius2342, 2017-10-19:
https://github.com/XKNX/xknx/tree/climate-test <-- i made a test without setpoint.

New setpoint shift is calculated out of old setpoint shift, old target temperature and desired new tart temperature.

Julius2342, 2018-01-07:
If there is anything missing, we should open a new issue.

closed 2018-01-07 by Julius2342