XKNX/xknx #15: group Address

twistermcneal, 2016-12-22:
the group address is in the wrong format!

the last bit is not used in the address

main  = 4bit
middle = 3bit
sub = 8bit
Address bit format:   0mmm msss gggg gggg

                 +-----------------------------------------------+
       16 bits   |                 GROUP ADDRESS                 |
                 +-----------------------+-----------------------+
                 | OCTET 0 (high byte)   |  OCTET 1 (low byte)   |
                 +--+--+--+--+--+--+--+--+--+--+--+--+--+--+--+--+
          bits   | 7| 6| 5| 4| 3| 2| 1| 0| 7| 6| 5| 4| 3| 2| 1| 0|
                 +--+--+--+--+--+--+--+--+--+--+--+--+--+--+--+--+
                 |  |  Main Group (S13)  |   Sub Group (S13)     |
                 +--+--------------------+-----------------------+

Julius2342, 2016-12-22:
die unittests müssten vermutlich auch noch angepasst werden.

Und weisst Du wie das bei physikalischen Addressen und bei Level2 Addressen ist?

twistermcneal, 2016-12-22:
Bei level 2 ist es wie oben in der Tabelle dargestellt  Main Group (S13)  |   Sub Group (S13)

physikalischen Addressen 
Format: Bereich [4bit] - Linie [4bit] - Busteilnehmer [1 Byte].

Julius2342, 2016-12-22:
ok, dann sollten wir da auch unterscheiden... :-/ danke für die Hilfe!

Julius2342, 2016-12-24:
Könntest du noch mal einen Blick auf meinen commit 9ed029378cc12d0d7b6564e244e0427d63ead23c werfen?

- ich habe zum einen die Unittests gefixt
- Ein- und Ausgabe von physikalischen Addressen ist jetzt weiter getrennt von Gruppenaddressen, da sie unterschiedliches Bitschema. (4-4-8 statt 5-3-8)
- Was mir noch unklar ist, ob das erste Bit in einer Gruppenaddresse benutzbar ist oder nicht, also sprich ob ein 31/7/255 zulässig ist. [Schulungsmaterialen](http://www.knx.org/fileadmin/template/documents/downloads_support_menu/KNX_tutor_seminar_page/basic_documentation/Communication_E1212a.pdf) lassen jedenfalls diesen Schluss zu. Woher hast Du Deine Information?

twistermcneal, 2016-12-25:
Sieht gut aus...
Du hast recht 31/7/255 ist verwendbar habe eine Routing Vorlage verwendet da wird das bit 7 zum Routing über zwei Linien verwendet...

Ich würde noch den Adressen typ versuchen zu bestimmen wenn nicht angegeben.
`
if address_type == None:
   if address.find(".") > 0 :
        self.address_type =  AddressType.PHYSICAL
    else:
        self.address_type = AddressType.GROUP  				
else:
    self.address_type = address_type		
`

Julius2342, 2016-12-25:
Gute Idee, bau ich ein ... sag mal, könntest du mal den branch cover-without-absolute-position-t5-improvements-twistermcneal testen und gucken ob Deine Änderungen funktionieren?

Verbringe leider Weihnachten im KH, wegen einer BlinddarmOP. Schlechtes Timing :-( Aber ich kann wenigstens basteln :-)

twistermcneal, 2016-12-25:
Ohje dann gleich mal gute Besserung...

Soweit ist es getestet ... es gibt aber noch einen Bug. Wenn ich von der max/min Position eine neue Position setze dann wird manchmal eine falsche aktuelle Position errechnet! 
Habs aber noch nicht gefunden.

Scheint aber immer dann zu passieren wenn man die Jalousie komplett auffährt ohne stop zu betätigen oder eine Position anzufahren...

Julius2342, 2016-12-25:
Danke :-) Ich bastel hier gerade schon an der Erkennung, aber die unit tests laufen noch nicht ...

Julius2342, 2016-12-28:
ich schließe das mal - und öffne einen neuen Bug mit dem von Dir beschriebenen Problem

closed 2016-12-28 by Julius2342