XKNX/xknx #28: Missing Clean Connect and Disconnect Procedure

daBONDi, 2016-12-29:
Missing Main Clean Connect and Disconnect Procedure to the Gateway

Realy required, Sessions keeping open on the gateway and on Gateways there are a maximum ammount
of sessions allowed!

daBONDi, 2017-02-01:
Need to Check the build out as soon as i can hopefully this weekend!

Julius2342, 2017-02-01:
Tunnelling is not yet working, for now i only added a full abstraction for necessary KNX/IP packets. What should work already is the gateway scanner.

Julius2342, 2017-02-21:
We now have a clean connect and disconnect procedure:

```
    tunnel = Tunnel(
        xknx,
        src_address,
        local_ip=gatewayscanner.found_local_ip,
        gateway_ip=gatewayscanner.found_ip_addr,
        gateway_port=gatewayscanner.found_port)

    yield from tunnel.connect_udp()
    yield from tunnel.connect()

    yield from tunnel.send_telegram(Telegram(Address('1/0/15'), payload=DPTBinary(0)))
    yield from asyncio.sleep(2)
    yield from tunnel.send_telegram(Telegram(Address('1/0/15'), payload=DPTBinary(1)))
    yield from asyncio.sleep(2)

    yield from tunnel.connectionstate()
    yield from tunnel.disconnect()
````

Julius2342, 2017-03-21:
yay, hass sends now a proper tunnel disconnect request to device when hass is shut down

Julius2342, 2017-07-23:
I consider this as done :-)

closed 2017-07-23 by Julius2342