XKNX/xknx #34: Polling Group Address

twistermcneal, 2017-03-26:
In a KNX system you dont have to poll group addresses .... each client sends is state if changed or periodically.
In a large system you will getting problems when you poll all groups

Julius2342, 2017-03-26:
See your point.

I found it pretty useful to do this at the beginning, to have the current state in HASS ...

Julius2342, 2017-03-26:
What about requesting all at the beginning (e.g. with lower throughput) and then only request group addresses which haven't been updated for a long time?

twistermcneal, 2017-03-26:
in my case this leads to bus heavy and i never get a answer ... yes i think thats a possibility

Julius2342, 2017-03-26:
i had a sleep within each request within the update loop, probably this got lost while refactoring

Julius2342, 2017-03-26:
may you increase the value of "sleep_during_devices" within stateupdater.py ?

twistermcneal, 2017-03-26:
think 0.5 is a good value i have 130 group addresses so it still runs :) but looks good

Julius2342, 2017-03-26:
May you make a PR for the increased values?

Julius2342, 2017-07-16:
with 0cee1e0d72882d7ea45d1463f1f7c426d5584195 the syncing is now synchronous, which means the next device is polled after the answer of the last device was received.

so no sleep during devices is needed.

Julius2342, 2017-07-16:
i also increased the time during complete refetches. It is now fetched once per hour.

closed 2017-07-16 by Julius2342