XKNX/xknx #46: Use python logging instead of print()

elupus, 2017-07-30:
The library should use python logging instead of print so that logging ends up correct format in using applications like home assistant.

Julius2342, 2017-07-30:
Yes :-) will add this as the next thing. It is also a duplicate of #26

elupus, 2017-07-30:
Whoops.. Sorry. My initial hass tests look good. Way more performant. Especially when you have sensors that don't reply as expected.

Julius2342, 2017-07-30:
If you have any ideas for improvement, tell me.

resoai, 2017-07-31:
HVAC support for XKNX Climate, please! This along with setpoint shifts will bring hass to the next level!

Julius2342, 2017-07-31:
@resoai : may you open a separate ticket for HVAC? I'm happy to implement this, but i need your help, bc i live in a country were AC is not needed :-)

My general plan is to add support for RTR (Room Temperature Regulation). The next step is to add valve support for my heating system. Would it be something similar for AC?

resoai, 2017-07-31:
To be honest I don't have A/C myself (same issue) but we do have heated floors which are being controlled via KNX. I'll be more than happy to test it and give you as much support as i can. Let's wait till PR is approved and merged and I'll start testing XKNX and shall create a separate ticket for HVAC :) I will be able to test HVAC with A/C on a remote house though!

elupus, 2017-08-03:
Think you can close this issue. Logging is using python logging now.

Julius2342, 2017-08-03:
Doing so!

closed 2017-08-03 by Julius2342