XKNX/xknx #48: Better climate support for HASS

Julius2342, 2017-08-01:
3/4 of all feedback i got wished better climate support. Looks like we need better climate support.

I suggest the following configuration:

```
climate:
   - platform: xknx
     name: Kitchen.Temperature
     temperature_address: '6/2/1'
     setpoint_address: '6/2/2'
     away_address: '6/2/3'
     operation_mode:
         - comfort
            action:
              - entity_id: switch.KitchenClimateComfort
                service: homeassistant.turn_on
              - entity_id: switch.KitchenClimateFrost
                service: homeassistant.turn_off
         - frost
            action:
              - entity_id: switch.KitchenClimateComfort
                service: homeassistant.turn_off
              - entity_id: switch.KitchenClimateFrost
                service: homeassistant.turn_on
         - normal
            action:
              - entity_id: switch.KitchenClimateComfort
                service: homeassistant.turn_off
              - entity_id: switch.KitchenClimateFrost
                service: homeassistant.turn_off
```

(and therefore add two [hidden switches for comfort mode and frost mode]).

The reason for this would be, that we are pretty open for future extensions within the module.

mag2bue, 2017-08-01:
for each mode -> comfort, frost and away, I have a seperate KNX adress

resoai, 2017-08-01:
http://www.knx.org/fileadmin/template/documents/downloads_support_menu/KNX_tutor_seminar_page/Advanced_documentation/Heating_E1211a.pdf

According to this document there are following operating modes:

- Frost/heat protection
- Comfort
- Night
- Standby

To be honest I would think that relevant controls  should simply appear if there is an "address" set in order to not over-complicate things. This would make a config for my place with 18 thermostats controls barely readable.

Julius2342, 2017-08-01:
hmmm.... :) you are about to convince me....

resoai, 2017-08-01:
To be honest I know very little about how thermostats work in KNX and am simply making assumptions :) Perhaps we could have a look at the implementation in OpenHAB? I'm sure they have it figured out.

Julius2342, 2017-08-01:
is there any knx representation of "away" ?

And do these modes exclude each other? (So can it be "night" and "comfort" in parallel ?)

Julius2342, 2017-08-01:
The problem i have, i do not fully understand this setpoint thing within KNX.

As far as i see it:

* There is a setpoint temperature, which is the temperature under normal conditions. (Sometimes also referred as "basis-setpoint?" )
* But then there are some "modifiers" like comfort which makes it warmer without changing the setpoint?

resoai, 2017-08-01:
Yes, they exclude each other, the mode can only be set to one of them. I could not find any mention of "away". Please have a look at the operating manual for a Jung room controller, maybe this would help.

[en_4093KRMTSD_V02_10052016_td.pdf](https://github.com/XKNX/xknx/files/1191812/en_4093KRMTSD_V02_10052016_td.pdf)

resoai, 2017-08-01:
Setpoint shift is basically the deviation (plus or minus) from the base temperature appropriate (set) for the particular operating mode. This is how I understand it.

Julius2342, 2017-08-01:
what a terrible concept ...

Julius2342, 2017-08-01:
so we should add group addresses for each state and add the exclusiveness-logic on top. Will do :)

resoai, 2017-08-01:
It makes a lot of sense actually. You can set the operating mode for the whole house yet you want to keep one room always colder. I know it feels very wrong at first and it took me some time to get used to it.

resoai, 2017-08-01:
Unfortunately I don't have ETS software but from I can see from the files left for me from the installer, the "operating mode"  is just one address.

Julius2342, 2017-08-01:
probably you can encode the modes also in several bits of the same byte

resoai, 2017-08-01:
While reading in the KNX manual they call it "Enumerated Types":

[05_Interworking_E1209.pdf](https://github.com/XKNX/xknx/files/1191851/05_Interworking_E1209.pdf)
```
Enumerations are used for datapoints with a clearly limited number of values and no
precise hierarchical order. If a datapoint uses an enumerated data type, all possible
states shall be described. If invalid / undefined values are sent, no malfunctions may
occur.
Examples are: Toggling between “Comfort”, “Standby” and “Night” operating modes in
room thermostats.
```

Perhaps the attached file could help (4.20 Common HVAC Datapoint types)

[05_Interworking_E1209.pdf](https://github.com/XKNX/xknx/files/1191852/05_Interworking_E1209.pdf)

mag2bue, 2017-08-02:
To manage it, we need  3 optional switches. (1bit  0 or 1)

 e.g. see page 45-47
German: http://www.mdt.de/download/MDT_THB_Heizungsaktor_02.pdf
English: http://www.mdt.de/download/MDT_TM_Heating_Actuator_02.pdf

In ETS, I need for each 1bit address to put ON/OFF
- comfort
- night
- frost
-> standby is a status of all -> OFF

Julius2342, 2017-08-02:
see my updated suggestion.

mag2bue, 2017-08-02:
perfect! :-)

mag2bue, 2017-08-02:
@Julius2342 

I have technical question:

drop-down: if I select e.g. night - > only "address of night" will be switched ON, all the other are to  OFF?
what is the technical background for "operating_mode" heating or cooling?

Or you control it by DPT HVAC Mode 20.102 ? see page 47/48

resoai, 2017-08-02:
Guys, this is the actuator you are talking about here, not the RTR which a climate component is. Climate component has to provide a level of abstraction, meaning that user sets a temperature/operating mode/setpoint shift and KNX figures the rest. I have a bunch of actuators for heated floors and KNX decides on the position of the valves etc.

mag2bue, 2017-08-02:
correct, totally I control already 16 heaters with OpenHab. The topic about we are talking it not to set the setpoint (This works already perfect in HA) I talking about, is to swich between the status. Like ON / OFF

resoai, 2017-08-02:
Setpoint temperature works but not setpoint shift :) I was under the impression that for heaters (ON/OFF) you just use a switch component.

mag2bue, 2017-08-02:
the setpoint switch e.g. modus "standby" -5k, "night -3k" you have to setup in ETS.

But we need a solution to change the modus ... in my case, I can setup this in ETS by a simple ON /OFF 1bit or HVAC Mode.

resoai, 2017-08-02:
Yes, changing modus (operating_mode) via HVAC mode is a must.

Julius2342, 2017-08-02:
Lets also add a group address for AC on/off and Heating on/off

resoai, 2017-08-02:
Surely we can do anything but it would be nice to perhaps stick with some sort of standard appropriate to KXN room controller?

mag2bue, 2017-08-02:
Standard is perfect. -> HVAC 
Sorry, maybe there was as well a misunderstanding from my site.

Looking forward to test the functions

resoai, 2017-08-02:
Perhaps I am also not (completely) right since I'm still trying to fully understand how KNX climate works and I don't actually have ETS software. One other (offtopic) cool thing would be to enable multiple group addresses for one switch/light. 

OpenHab's implementation is messy yet very flexible.

http://docs.openhab.org/addons/bindings/knx1/readme.html

Julius2342, 2017-08-02:
@resoai : KNX itself is just a bus (without any active logic). The control of the valves/etc. is done within climate devices (or thermostats). I for example have GIRA Touch Sensor 3 plus: http://www.gira.com/produkte/abbildungen/zoom1/001223_1384961308.jpg

My original idea was to be open to control the valves via hass itself - e.g. via a complete new component. This then could be much smarter than the small devices on the wall. But the current approach of syncing the flags/values with the climate control, we do at least nothing wrong.

Julius2342, 2017-08-08:
@mag2bue : I need your help.

Operation Modes
-----------------

Did i get this right:

* so you have binary addresses for comfort, frost and away.
* You don't have an dedicated address for standby (i assume this is just all other states switched off).
* For switching to away, i would switch off frost, comfort and switch on away.
* For switching to frost, i would switch off away, comfort and switch on frost.
* For switching to comfort, i would switch off away, frost and switch on comfort.
* For switching to standby i would switch off away, frost and comfort.

Setpoint/Target temperature
-----------------------------

Do you (like i have) have also two addresses:

* basis set point address. Read/Write for setting the set point in comfort mode.
* target temperature. Read only, calculated from basis set point and operation mode.

mag2bue, 2017-08-08:
@ulius2342 

Operation Modes as specified by you.

Setpoint, like today in HAAS for KNX

NEW: nice to have: to see in percent the control value   (Stellwert Venitl)  :-)

resoai, 2017-08-08:
Please do not forget about setpoint shift. I've also noticed that if you do not specify setpoint_temperature at all than the temperature control is still visible in the popup and shows random values if you click up/down.

Julius2342, 2017-08-08:
my approach is:

- i can read target-temperature and setpoint from KNX bus and therefore can calculate the shift.
- within hass i set the target-temperature. calculate the new setpoint and set it.
- when changing operation mode, the target temperature changes.

resoai, 2017-08-08:
Sounds reasonable! There should be limits perhaps, since shift can not go above and bellow certain values.

resoai, 2017-08-08:
I've checked and it looks like I have a special address for setpoint temperature but it is used for feedback (to be displayed on the room controller) i.e. read-only. It updates correctly in hass if I push physical buttons but not the other way around.

Julius2342, 2017-08-08:
yes! thats how it is here.

* base setpoint: read/write
* set/point: read only. (the target temperature at the moment.)

Example:

* base setpoint = 21
* setpoint = 22 (bc. you are in comfort mode)

within hass i would display 22. bc. this is the desired temperature at the moment. if you change the target temperature, e.g. to 21.5. I do the following:

```
    @asyncio.coroutine
    def set_target_temperature(self, setpoint):
        """calculate setpoint-delta and setpoint and send it to  KNX bus."""
        setpoint_delta = self.setpoint - self.target_temperature
        new_setpoint = setpoint + setpoint_delta
        yield from self.set_setpoint(new_setpoint)
```

-> I sent 20.5 as new basis setpoint to the bus.

resoai, 2017-08-08:
Great! Looking forward to testing it!

Julius2342, 2017-08-09:
@mag2bue @resoai : I pushed a version. May I ask you to test? (XKNX also has to be updated to version 0.7.6.

@mag2bue : I could not test the mode with different binary addresses for night/comfort/frost. There is a chance, that it is not working as expected.

resoai, 2017-08-09:
Sadly I'm only able to test it remotely as I'm in the office but I've enabled one controller and here is what I see:

 - Current and target temperature are being displayed correctly 
 - Operation mode changes correctly
 - The initial state of the operation mode (after restart) is not being read correctly, it displayed standby when it was comfort.
 - Setpoint doesn't work, it does not change the shift and target address. 

Also, now the following error appeared:

```
2017-08-09 10:46:53 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/data/srv/homeassistant/lib/python3.5/site-packages/homeassistant/core.py", line 1025, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/data/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/climate/__init__.py", line 321, in async_temperature_set_service
    yield from climate.async_set_temperature(**kwargs)
  File "/data/srv/homeassistant/config/custom_components/climate/xknx.py", line 149, in async_set_temperature
    yield from self.device.set_target_temperature(temperature)
  File "/data/srv/homeassistant/lib/python3.5/site-packages/xknx/devices/climate.py", line 161, in set_target_temperature
    setpoint_delta = self.setpoint - self.target_temperature
TypeError: unsupported operand type(s) for -: 'NoneType' and 'float'
```

resoai, 2017-08-09:
May  I also suggest following, purely for cosmetic purposes:

- In climate popup to charge "operation" to "operation mode"
- operation_mode_protection_address to change to operation_mode_frost_protection_address (simply protect sounds weird)

resoai, 2017-08-09:
We currently have a barely known nomos system to control KNX in our house. This is how they (poorly) present climate. I'm thinking, perhaps we should do the same with setpoint shift just to keep it standard? You can also see how heating demand (valve position) is being displayed. I think it would be able to display this along with A/C status and intensity. We don't have A/C in the UK but I can test it remotely in our flat in Ukraine.  

[image]

Julius2342, 2017-08-09:
I also want to see the valve positions :-) at the moment i do this with a sensor:

```yaml
sensor:
  - platform: xknx
    name: Hass-Heating.Valve1
    address: '2/0/0'
    type: 'percent'
```

This works more-or-less but is not nice. My mid-term plan is to introduce a valve component within HASS - which could also be used to control irrigation in the garden.

The problem with setpoint shift is, that this value is not present within the climate card. so we would have to introduce our own setcard. But have no idea how this templating system works.

Where in the Ukraine? Im often in Lviv.

resoai, 2017-08-09:
It would be nice to have our own template. Oh, wow! I'm from Odessa but have recently moved to London. I travel back to Odessa twice a year.

resoai, 2017-08-09:
Shouldn't we be sending setpoint shift i.e. -0.5K ... 1.5K to KNX Bus?

Julius2342, 2017-08-13:
@resoai : how does setpoint shift work? (it seems that this functionality is not supported by my devices)

mag2bue, 2017-08-13:
@resoai  no, you have one main temeperature.  (basis value) Depend on you status, you will have you will have your shift -> 4k

FireFrei, 2017-08-13:
I just pulled the newest xknx and hass version. On startup of hass i receive the following error:

```
2017-08-13 17:29:11 ERROR (MainThread) [homeassistant.components.climate] Error while setting up platform xknx
Traceback (most recent call last):
  File "/usr/src/app/homeassistant/helpers/entity_component.py", line 164, in _async_setup_platform
    SLOW_SETUP_MAX_WAIT, loop=self.hass.loop)
  File "/usr/local/lib/python3.6/asyncio/tasks.py", line 358, in wait_for
    return fut.result()
  File "/usr/local/lib/python3.6/asyncio/futures.py", line 245, in result
    raise self._exception
  File "/usr/local/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/config/custom_components/climate/xknx.py", line 49, in async_setup_platform
    yield from add_devices_from_platform(hass, config, add_devices)
  File "/usr/local/lib/python3.6/asyncio/coroutines.py", line 210, in coro
    res = func(*args, **kw)
  File "/config/custom_components/climate/xknx.py", line 86, in add_devices_from_platform
    CONF_OPERATION_MODE_COMFORT_ADDRESS))
TypeError: __init__() got an unexpected keyword argument 'group_address_target_temperature'
```

HASS 0.51.1
xknx 0.7.8

Config:

```
climate:
  - platform: xknx
    name: 'eg_wohnzimmer_rtr01'
    temperature_address: 1111
    setpoint_address: 2222
    target_temperature_address: 3333
    operation_mode_protection_address: 4444
    operation_mode_night_address: 5555
    operation_mode_comfort_address: 6666
```

Julius2342, 2017-08-13:
are you sure that you have the current xknx library installed? (bc xknx.Climate has a keyword argument group_address_target_temperature ).

FireFrei, 2017-08-13:
Okay, stupid PIP again.  Removing everything manually and installing it again (instead of pip install --upgrade xknx) solved it...

Here are my testing results:
Reading current temp and setpoint temp works. Updating setpoint temp from hass results in:
```
2017-08-13 17:57:21 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/usr/src/app/homeassistant/core.py", line 1025, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/usr/src/app/homeassistant/components/climate/__init__.py", line 325, in async_temperature_set_service
    yield from climate.async_set_temperature(**kwargs)
  File "/config/custom_components/climate/xknx.py", line 149, in async_set_temperature
    yield from self.device.set_target_temperature(temperature)
  File "/config/deps/lib/python3.6/site-packages/xknx/devices/climate.py", line 161, in set_target_temperature
    setpoint_delta = self.setpoint - self.target_temperature
TypeError: unsupported operand type(s) for -: 'NoneType' and 'float'
```

Changing operation mode works when selecting the mode for a single entity. If I have a group of climate controls (xknx and others) I get the following error when selecting a mode, e.g. comfort:
```
2017-08-13 17:55:07 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/usr/src/app/homeassistant/core.py", line 1025, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/usr/src/app/homeassistant/components/climate/__init__.py", line 376, in async_operation_set_service
    yield from climate.async_set_operation_mode(operation_mode)
  File "/config/custom_components/climate/xknx.py", line 170, in async_set_operation_mode
    xknx_operation_mode = ClimateOperationMode(operation_mode)
  File "/usr/local/lib/python3.6/enum.py", line 291, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.6/enum.py", line 533, in __new__
    return cls._missing_(value)
  File "/usr/local/lib/python3.6/enum.py", line 546, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 'comfort' is not a valid ClimateOperationMode
```

resoai, 2017-08-13:
Thanks for the PR, I've just updated to it. I will have to speak to the guy who installed KNX in our house but I'm mostly certain that setpoint shift is being sent to the KNX bus. As of now there are two major problems:

- setpoint temperature can't be set (as I mentioned above) and I get following error when I try changing the temerature:

```
2017-08-13 18:03:30 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/data/srv/homeassistant/lib/python3.5/site-packages/homeassistant/core.py", line 1025, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/data/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/climate/__init__.py", line 325, in async_temperature_set_service
    yield from climate.async_set_temperature(**kwargs)
  File "/data/srv/homeassistant/config/custom_components/climate/xknx.py", line 149, in async_set_temperature
    yield from self.device.set_target_temperature(temperature)
  File "/data/srv/homeassistant/lib/python3.5/site-packages/xknx/devices/climate.py", line 161, in set_target_temperature
    setpoint_delta = self.setpoint - self.target_temperature
TypeError: unsupported operand type(s) for -: 'NoneType' and 'float'
2017-08-13 18:03:41 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/data/srv/homeassistant/lib/python3.5/site-packages/homeassistant/core.py", line 1025, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/data/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/climate/__init__.py", line 325, in async_temperature_set_service
    yield from climate.async_set_temperature(**kwargs)
  File "/data/srv/homeassistant/config/custom_components/climate/xknx.py", line 149, in async_set_temperature
    yield from self.device.set_target_temperature(temperature)
  File "/data/srv/homeassistant/lib/python3.5/site-packages/xknx/devices/climate.py", line 161, in set_target_temperature
    setpoint_delta = self.setpoint - self.target_temperature
TypeError: unsupported operand type(s) for -: 'NoneType' and 'float'
2017-08-13 18:05:14 ERROR (Thread-33) [pychromecast.socket_client] Failed to connect, retrying in 5.0s
```

- initial state (after restart of hass) of the operation mode is incorrect (it shows standby for all controllers, when they are in comfort mode).

I have also received following message on startup:

```
2017-08-13 18:02:53 ERROR (MainThread) [homeassistant.components.notify] Failed to initialize notification service xknx
```

resoai, 2017-08-13:
Perhaps there are two different approaches, like setpoint temperature and setpoint shift?

[en_2178_td.pdf](https://github.com/XKNX/xknx/files/1220608/en_2178_td.pdf)

mag2bue, 2017-08-13:
setpoint should be 2bytes value... sorry, currently I have no time for testing :-( mybe next week!

resoai, 2017-08-13:
Setpoint temperature yes, 2 bytes object, but setpoint shift is 1 byte:

```
Preset setpoint shift
1-byte object for presetting a basic setpoint shift, e.g. via a controller
extension.
x ≤ 0 ≤ y (0 = no active shifting); integers
The possible range of values (x to y) is fixed by the setting of the upper
and lower limits for the setpoint (parameterizable) in combination with
the step value (0.5 °C). 
```

Julius2342, 2017-08-13:
@resoai : hmm.  :-)

Julius2342, 2017-08-13:
@FireFrei : looks like the calculating of the temperature has to be more failure-proof. Will think about this on Tuesday.

Regarding the message on startup within notify. You can ignore this, this only happens when installing it as custom component.

resoai, 2017-08-16:
Sorry to bother you but the climate control is broken for me now:

1. operation mode is set incorrectly on startup
2. setpoint temperature can not be set (because of the reasons I mentioned before).

Julius2342, 2017-08-16:
Hi @resoai : lets try to debug/fix this in the next couple of days.

resoai, 2017-08-16:
Certainly! I'm here to test!

Kind regards,
Alexey Ivanov

TinusK, 2017-09-06:
Setting HVAC modes work very well as expected.
Only thing missing is Setpoint Offset. Can I help with that?

TinusK, 2017-09-06:
Setpoint shift solution could look like this:
setpoint_stepsize
setpoint_stepadjust_address (1 Byte)
target_temperature_address

For example:
setpoint_stepsize=0.5 (should be the same as set in the KNX Heating Aktuator)
current target_temperature_address=21C
Clicking 6 times on the '+' in HAss would produce 21+(6x0.5)=24 and sends on KNX bus setpoint_stepadjust_address: 6
The KNX bus reports the new target_temperature_address back: 24C

Clicking 3 times on the '-' in HAss would produce 24-(3x0.5)=22.5 and sends on the KNX bus setpoint_stepadjust_address: 3
The KNX bus reports the new target_temperature_address back: 22.5

Julius2342, 2017-09-06:
Thank you! My plan is to implement it this afternoon. I will com back if I need help!

[60 more replies]

closed 2017-11-05 by Julius2342