XKNX/xknx #49: Errors on startup

Julius2342, 2017-08-02:
Background
------------

@elupus elupus discovered some errors on initial startup.

Plan to reproduce
------------------

Do

```
 rm -r ~/.homeassistant/deps/lib/python/site-packages/xknx*
``` 

and check wether the error re-occurs.

elupus, 2017-08-03:
Sorry for the cut log, but here is the error on start.

```
2017-08-03 10:26:42 INFO (MainThread) [homeassistant.loader] Loaded binary_sensor from homeassistant.components.binary_sensor
2017-08-03 10:26:42 ERROR (MainThread) [homeassistant.loader] Error loading homeassistant.components.binary_sensor.xknx. Make sure all dependencies are installed
Traceback (most recent call last):
  File "/home/homeassistant/home-assistant/homeassistant/loader.py", line 142, in get_component
    module = importlib.import_module(path)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 673, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/homeassistant/home-assistant/homeassistant/components/binary_sensor/xknx.py", line 113, in <module>
    class XKNXBinarySensor(BinarySensorDevice):
  File "/home/homeassistant/home-assistant/homeassistant/components/binary_sensor/xknx.py", line 116, in XKNXBinarySensor
    import xknx
ImportError: No module named 'xknx'
2017-08-03 10:26:42 ERROR (MainThread) [homeassistant.loader] Unable to find component binary_sensor.xknx
2017-08-03 10:26:42 ERROR (MainThread) [homeassistant.loader] Error loading homeassistant.components.binary_sensor.xknx. Make sure all dependencies are installed
Traceback (most recent call last):
  File "/home/homeassistant/home-assistant/homeassistant/loader.py", line 142, in get_component
    module = importlib.import_module(path)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 673, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/homeassistant/home-assistant/homeassistant/components/binary_sensor/xknx.py", line 113, in <module>
    class XKNXBinarySensor(BinarySensorDevice):
  File "/home/homeassistant/home-assistant/homeassistant/components/binary_sensor/xknx.py", line 116, in XKNXBinarySensor
    import xknx
ImportError: No module named 'xknx'
2017-08-03 10:26:42 ERROR (MainThread) [homeassistant.loader] Unable to find component binary_sensor.xknx
2017-08-03 10:26:42 INFO (MainThread) [homeassistant.setup] Setting up binary_sensor

```

elupus, 2017-08-03:
Okey. Think i found it. The binary sensor does import xknx, but doesn't state xknx as a requirement.

elupus, 2017-08-03:
Seems you already solved the issue with the latest changes you did :). So feel free to close.

Julius2342, 2017-08-03:
Yes, hound complained about it :) Closing this issue.

closed 2017-08-03 by Julius2342