XKNX/xknx #50: Timout problems on large KNX environments

Julius2342, 2017-08-02:
Background
------------

@elupus experienced the following exception during sync:

```
Jul 31 04:55:05 hass hass[51396]: Sync Uttag Kök
Jul 31 04:55:05 hass hass[51396]: Exception while processing telegram: Could not send telegram to tunnel
Jul 31 04:55:05 hass hass[51396]: Traceback (most recent call last):
Jul 31 04:55:05 hass hass[51396]:   File "/home/homeassistant/evn_dev/lib/python3.5/site-packages/xknx/core/telegram_queue.py", line 79, in process_telegram
Jul 31 04:55:05 hass hass[51396]:     yield from self.process_telegram_outgoing(telegram)
Jul 31 04:55:05 hass hass[51396]:   File "/home/homeassistant/evn_dev/lib/python3.5/site-packages/xknx/core/telegram_queue.py", line 91, in process_telegram_outgoing
Jul 31 04:55:05 hass hass[51396]:     yield from self.xknx.knxip_interface.send_telegram(telegram)
Jul 31 04:55:05 hass hass[51396]:   File "/home/homeassistant/evn_dev/lib/python3.5/site-packages/xknx/io/knxip_interface.py", line 139, in send_telegram
Jul 31 04:55:05 hass hass[51396]:     yield from self.interface.send_telegram(telegram)
Jul 31 04:55:05 hass hass[51396]:   File "/home/homeassistant/evn_dev/lib/python3.5/site-packages/xknx/io/tunnel.py", line 111, in send_telegram
Jul 31 04:55:05 hass hass[51396]:     raise Exception("Could not send telegram to tunnel")
Jul 31 04:55:05 hass hass[51396]: Exception: Could not send telegram to tunnel
````

This error occurs when the `xknx.io.Tunnelling` sends `xknx.knxip.TunnellingRequest` but did not receive an `xknx.knxip.TunnellingAck` within 1 second.

Open Questions
----------------

* When does this happen? At startup? When Stateupdater does its regular sync?
* How often does this happen?
* We should find out if the bus always tries to read the same packets
* Do we get any KNX/IP feedback from the router?
* Does the Tunnel repair itself after ~1 minute? The tunnel abstraction sends a StateRequest every 15 seconds and restarts the tunneling after the 3rd failed attempt.

Suggested approaches for mitigation
-------------------------------------

There are several ideas of mitigating this:

* **RequestResponse** / **ValueReader** : Increase the timeout for RequestResponse and for ValueReader (and make this timeout configurable)
* **KNXIPInterface** : Directly send telegrams to the bus without using TelegramQueue
* **StateUpdater** : Only sync status when telegram qeuue is empty. And/Or do a short sleep between device.sync()s.

elupus, 2017-08-03:
Its likely not a timeout issue. I would expect it's a dropped connection from the server side. xknx doesn't implemented the following retry mechanism from spec:

```
If a TUNNELLING_REQUEST frame is not confirmed within the TUNNELLING_REQUEST_TIME_- OUT time of one (1) second then the frame shall be repeated once with the same sequence counter value by the sending KNXnet/IP device.

If the KNXnet/IP device does not receive a TUNNELLING_ACK frame within the TUNNELLING_- REQUEST_TIMEOUT (= 1 second) or the status of a received TUNNELLING_ACK frame signals any kind of error condition, the sending device shall repeat the TUNNELLING_REQUEST frame once and then terminate the connection by sending a DISCONNECT_REQUEST frame to the other device’s control endpoint.
```

Julius2342, 2017-08-03:
hmm, you are right. This kind of procedure is not (yet) covered.

But: We should also investigate, why this scenario is happening on your setup and avoid this by mitigating bottle necks.

elupus, 2017-08-03:
I have a 3g mobile connection between knx installation and hass. So not too
unlikely to happen.

Julius2342, 2017-08-04:
May you just for a test increase the RequestResponse timeout:

```
diff --git a/xknx/io/request_response.py b/xknx/io/request_response.py
index 11ca606..a271113 100644
--- a/xknx/io/request_response.py
+++ b/xknx/io/request_response.py
@@ -11,7 +11,7 @@ class RequestResponse():
 
     # pylint: disable=too-many-instance-attributes
 
-    def __init__(self, xknx, udp_client, awaited_response_class, timeout_in_seconds=1):
+    def __init__(self, xknx, udp_client, awaited_response_class, timeout_in_seconds=5):
         """Initialize RequstResponse class."""
         self.xknx = xknx
         self.udpclient = udp_client
```

Julius2342, 2017-08-04:
Got it reproduced locally with the following patch:

```
diff --git a/xknx/io/udp_client.py b/xknx/io/udp_client.py
index 0c5cf77..473a1a6 100644
--- a/xknx/io/udp_client.py
+++ b/xknx/io/udp_client.py
@@ -7,6 +7,9 @@ Due to lame support of UDP multicast within asyncio some special treatment for m
 import asyncio
 import socket
 
+from random import randint
+from xknx.knxip import TunnellingAck, TunnellingRequest
+
 from xknx.knxip import KNXIPFrame
 from xknx.exceptions import CouldNotParseKNXIP, XKNXException
 
@@ -84,7 +87,12 @@ class UDPClient:
                 knxipframe = KNXIPFrame()
                 knxipframe.from_knx(raw)
                 self.xknx.knx_logger.debug("Received: %s", knxipframe)
-                self.handle_knxipframe(knxipframe)
+                if isinstance(knxipframe.body, TunnellingAck) and randint(0, 9) == 0:
+                    print("DROPPING PACKAGE")
+                else:
+                    self.handle_knxipframe(knxipframe)
             except CouldNotParseKNXIP as couldnotparseknxip:
                 self.xknx.logger.exception(couldnotparseknxip)
 
@@ -176,6 +184,10 @@ class UDPClient:
         if self.transport is None:
             raise XKNXException("Transport not connected")
 
+        if isinstance(knxipframe.body, TunnellingRequest) and randint(0, 9) == 0:
+            print("DROPPING PACKAGE SEND")
+            return
+
         if self.multicast:
             self.transport.sendto(bytes(knxipframe.to_knx()), self.remote_addr)
         else:
```

**So the problem is not a timeout but a dropped send:**

If you send an UDP packet which was never received by the router, XKNX tries to send it again but the tunnel abstraction has already increased the sequence number. XKNX and router get out of sync.

This condition is not recovered by the tunnel/heartbeat mechanism, bc the router still will reply correctly to connection state requests (from his perspective everything is fine...)

elupus, 2017-08-04:
Testing in progress. It usually breaks down in a day or so for me.. I'm a bit surprised thou that your test worked. I don't think there is anything in the specification that disallows skips in the sequence counter, but maybe there is.

elupus, 2017-08-05:
So far so good. Still connected.

Julius2342, 2017-08-05:
Yay :)

elupus, 2017-08-07:
Think you can close this as resolved. Still up and running.

Julius2342, 2017-08-07:
Yay :)

closed 2017-08-07 by Julius2342