XKNX/xknx #51: auto_update cover

twistermcneal, 2017-08-03:
2017-08-03 17:40:13 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.6/asyncio/tasks.py", line 179, in _step
    result = coro.send(None)
  File "/usr/lib/python3.6/site-packages/homeassistant/core.py", line 1025, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/usr/lib/python3.6/site-packages/homeassistant/components/cover/__init__.py", line 173, in async_handle_cover_service
    yield from getattr(cover, method['method'])(**params)
  File "/config/custom_components/cover/xknx.py", line 128, in async_close_cover
    self.start_auto_updater()
  File "/config/custom_components/cover/xknx.py", line 154, in start_auto_updater
    self.hass, self.auto_updater_hook)
  File "/usr/lib/python3.6/site-packages/homeassistant/helpers/event.py", line 26, in factory
    hass.loop, ft.partial(async_factory, *args, **kwargs)).result()
  File "/usr/lib/python3.6/site-packages/homeassistant/util/async.py", line 160, in run_callback_threadsafe
    raise RuntimeError('Cannot be called from within the event loop')
RuntimeError: Cannot be called from within the event loop

I think we have this error since asyncio implementation

Results in no updates for the cover and no position change from the traveling calculater

Julius2342, 2017-08-03:
looks like an `asyncio` problem. Will have a look.

Julius2342, 2017-08-03:
Looks like the problem is in here:

```
    def auto_updater_hook(self, now):
        """Callback for autoupdater."""
        # pylint: disable=unused-argument
        self.schedule_update_ha_state()  # <---- cant be executed within asyncio
        if self.device.position_reached():
            self.stop_auto_updater()
```

Julius2342, 2017-08-03:
May I ask you to try out the following patch:

```
diff --git a/homeassistant/components/cover/xknx.py b/homeassistant/components/cover/xknx.py
index bd17f69..f6c6751 100644
--- a/homeassistant/components/cover/xknx.py
+++ b/homeassistant/components/cover/xknx.py
@@ -8,7 +8,8 @@ import asyncio
 import voluptuous as vol
 
 from homeassistant.components.xknx import DATA_XKNX, _LOGGER
-from homeassistant.helpers.event import track_utc_time_change
+from homeassistant.helpers.event import async_track_utc_time_change
+from homeassistant.core import callback
 from homeassistant.components.cover import PLATFORM_SCHEMA, CoverDevice
 from homeassistant.const import CONF_NAME
 import homeassistant.helpers.config_validation as cv
@@ -155,7 +156,7 @@ class XKNXCover(CoverDevice):
     def start_auto_updater(self):
         """Start the autoupdater to update HASS while cover is moving."""
         if self._unsubscribe_auto_updater is None:
-            self._unsubscribe_auto_updater = track_utc_time_change(
+            self._unsubscribe_auto_updater = async_track_utc_time_change(
                 self.hass, self.auto_updater_hook)
 
     def stop_auto_updater(self):
@@ -164,10 +165,11 @@ class XKNXCover(CoverDevice):
             self._unsubscribe_auto_updater()
             self._unsubscribe_auto_updater = None
 
+    @callback
     def auto_updater_hook(self, now):
         """Callback for autoupdater."""
         # pylint: disable=unused-argument
-        self.schedule_update_ha_state()
+        self.hass.async_add_job(self.async_update_ha_state())
         if self.device.position_reached():
             self.stop_auto_updater()
```

It is just a shot in the dark ...

twistermcneal, 2017-08-03:
Monsterkill !!!
i think you can apply the patch!!

Julius2342, 2017-08-03:
Yay :-) Any other feedback appreciated!

closed 2017-08-03 by Julius2342