XKNX/xknx #52: Raw access to KNX bus from HASS module

Julius2342, 2017-08-05:
Background
------------

@ FireFrei  suggested to add raw access to  KNX bus from Home Assistant Plugin:

> Is it possible to use xknx to access "raw knx messages" in automation like you can do in the current knx component?
> Example: You can create an automation that listens on the event "knx_event" which includes all data received on the knx bus.
> This makes it possible to use "special" knx devices that aren't implemented yet. In my case: I have some wall buttons - when I press them, i'd like to run an automation.

> Aaaand what about the other way around: Is it possible to send "raw" messages to the knx bus using a HASS service?

Implementation
---------------

```
xknx:
    fire_event: True
```

* ***fire_event** (Optional) if set to True, platform will write all received KNX messages to event bus

Additionally you can now use the following service:

Domain: xknx
Service: send 
Service Data: {"address": "1/0/15", "payload": 0}

* ***address*** : KNX group addresss
* ***payload*** : payload, either an integer or an array of integers

Julius2342, 2017-08-05:
I discovered a problem that internally XKNX parses the destination addresses within CEMI messages in the wrong way. Group addresses are displayed as physical addresses (1.2.3) instead of group addresses (1/2/3). Working on it 👍

FireFrei, 2017-08-05:
Yeah! cool!

Maybe it's also helpful to whitlist group addresses to listen to, especially for large knx setups. Like requested from @pvizeli in https://github.com/home-assistant/home-assistant/pull/8449   :

> Do not route all knx message to event bus. We remove this code in future. Please add a whitlist filter of address they you want see for things like automation in the event bus. All other is bad for our performance. Thanks

Julius2342, 2017-08-07:
@FireFrei : How would you expect such a filter? Just a list of addresses or some wildcards :-)

FireFrei, 2017-08-07:
@Julius2342 A list of addresses would solve the issue and prevent the hass core from too many 'useless' knx messages. 
I think most will use the "knx listener" for single devices or a group of the same devices. So a list with hardcoded addresses should actually work.
If it's possible to add both, a list with entries as group addresses and wildcards for group addresses, this would be a great improvement of usability in large knx setups.

Julius2342, 2017-08-13:
@FireFrei @pvizeli : Added filter functionality:

```
xknx:
    fire_event: True
    fire_event_filter: ["1/0/*", "6/2,3,4-6/*"]
```

closed 2017-08-13 by Julius2342