XKNX/xknx #55: Gateway scanner does not work with eibPort3

Julius2342, 2017-09-04:
@TinusK reported that the gateway scanner does not work with eibPort3 ( http://bab-tec.de/index.php/eibport_v3_de.html ).

TinusK, 2017-09-04:
How can I best report to you about things that work or don't work before opening an issue (could be something I'm doing wrong and it's not an issue - if you get my drift :))

For example:
on [http://xknx.io/home_assistant](url) it's said that switches need an `group_address` but that is wrong, it needs `address`. Binary sensors are working. Normal sensors not, but maybe I'm doing something wrong and are there more sensor types, eg. illuminance or speed?

Julius2342, 2017-09-04:
uh, yes ...  http://xknx.io/home_assistant is completely outdated. Do you use Skype or any other chat tool?

TinusK, 2017-09-04:
Skype would be fine.

Julius2342, 2017-09-04:
My Skype user name equals the github username :)

Julius2342, 2017-09-05:
Could you try the following:

Start tcpdump with the following parameters: `sudo tcpdump port 3671 -X -i enp3s0` (Replace the interface with the interface connecting your KNX/IP device).

Then start the gateway scanner with the following command: `python3 examples/example_gatewayscanner.py` .

The interesting part would be if you see on or two packets (the first is the request, the second the response). I see the following: 

```
   julius@walli ~:$ sudo tcpdump port 3671 -X -i enp3s0 
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on enp3s0, link-type EN10MB (Ethernet), capture size 262144 bytes
08:41:47.556799 IP 192.168.42.1.45813 > 224.0.23.12.3671: UDP, length 14
	0x0000:  4500 002a 7136 4000 0211 25d7 c0a8 2a01  E..*q6@...%...*.
	0x0010:  e000 170c b2f5 0e57 0016 e1dd 0610 0201  .......W........
	0x0020:  000e 0801 c0a8 2a01 b2f5                 ......*...
08:41:47.557747 IP 192.168.42.10.3671 > 192.168.42.1.45813: UDP, length 80
	0x0000:  4500 006c 0000 4000 8011 2525 c0a8 2a0a  E..l..@...%%..*.
	0x0010:  c0a8 2a01 0e57 b2f5 0058 de95 0610 0202  ..*..W...X......
	0x0020:  0050 0801 c0a8 2a0a 0e57 3601 0200 1100  .P....*..W6.....
	0x0030:  0000 0008 2d40 42dc e000 170c 000a b327  ....-@B........'
	0x0040:  26eb 4769 7261 204b 4e58 2f49 502d 526f  &.Gira.KNX/IP-Ro
	0x0050:  7574 6572 0000 0000 0000 0000 0000 0000  uter............
	0x0060:  0c02 0201 0302 0401 0501 0701            ............
```

TinusK, 2017-09-05:
I first started example_gatewayscanner.py without the tcpdump and it returned with `No gateway found`. The second try however, it did:
`Gateway found: epv3 KNXNet/IP / 192.168.1.22:3671
- Device supports tunneling
- Device supports routing, connecting via 192.168.1.21`

The third try (this time with tcpdump):
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
07:13:16.602478 IP epv3le-gm.3671 > hassio.40680: UDP, length 23
        0x0000:  4500 0033 0000 4000 4011 b73e c0a8 0116  E..3..@.@..>....
        0x0010:  c0a8 0115 0e57 9ee8 001f d6b5 0610 0420  .....W..........
        0x0020:  0017 0402 3f00 2900 bce0 1214 0700 0300  ....?.).........
        0x0030:  8000 29                                  ..)
07:13:16.602772 IP epv3le-gm.3671 > hassio.49189: UDP, length 23
        0x0000:  4500 0033 0000 4000 4011 b73e c0a8 0116  E..3..@.@..>....
        0x0010:  c0a8 0115 0e57 c025 001f ba79 0610 0420  .....W.%...y....
        0x0020:  0017 0401 3a00 2900 bce0 1214 0700 0300  ....:.).........
        0x0030:  8000 29                                  ..)
07:13:16.604590 IP hassio.40680 > epv3le-gm.3671: UDP, length 10
        0x0000:  4500 0026 0736 4000 4011 b015 c0a8 0115  E..&.6@.@.......
        0x0010:  c0a8 0116 9ee8 0e57 0012 839f 0610 0421  .......W.......!
        0x0020:  000a 0402 3f00                           ....?.
07:13:16.611931 IP hassio.49189 > epv3le-gm.3671: UDP, length 10
        0x0000:  4500 0026 0737 4000 4011 b014 c0a8 0115  E..&.7@.@.......
        0x0010:  c0a8 0116 c025 0e57 0012 839f 0610 0421  .....%.W.......!
        0x0020:  000a 0401 3a00                           ....:.
07:13:17.252153 IP epv3le-gm.3671 > hassio.40680: UDP, length 23
        0x0000:  4500 0033 0000 4000 4011 b73e c0a8 0116  E..3..@.@..>....
        0x0010:  c0a8 0115 0e57 9ee8 001f 3114 0610 0420  .....W....1.....
        0x0020:  0017 0402 4000 2900 bce0 127c 1832 0300  ....@.)....|.2..
        0x0030:  8007 bc                                  ...
07:13:17.252468 IP epv3le-gm.3671 > hassio.49189: UDP, length 23
        0x0000:  4500 0033 0000 4000 4011 b73e c0a8 0116  E..3..@.@..>....
        0x0010:  c0a8 0115 0e57 c025 001f 14d8 0610 0420  .....W.%........
        0x0020:  0017 0401 3b00 2900 bce0 127c 1832 0300  ....;.)....|.2..
        0x0030:  8007 bc                                  ...
07:13:17.255071 IP hassio.40680 > epv3le-gm.3671: UDP, length 10
        0x0000:  4500 0026 0753 4000 4011 aff8 c0a8 0115  E..&.S@.@.......
        0x0010:  c0a8 0116 9ee8 0e57 0012 839f 0610 0421  .......W.......!
        0x0020:  000a 0402 4000                           ....@.
07:13:17.258448 IP hassio.49189 > epv3le-gm.3671: UDP, length 10
        0x0000:  4500 0026 0754 4000 4011 aff7 c0a8 0115  E..&.T@.@.......
        0x0010:  c0a8 0116 c025 0e57 0012 839f 0610 0421  .....%.W.......!
        0x0020:  000a 0401 3b00                           ....;.
07:13:17.271621 IP epv3le-gm.3671 > hassio.40680: UDP, length 21
        0x0000:  4500 0031 0000 4000 4011 b740 c0a8 0116  E..1..@.@..@....
        0x0010:  c0a8 0115 0e57 9ee8 001d e69e 0610 0420  .....W..........
        0x0020:  0015 0402 4100 2900 bce0 12ff 1f32 0100  ....A.)......2..
        0x0030:  80                                       .
07:13:17.271923 IP epv3le-gm.3671 > hassio.49189: UDP, length 21
        0x0000:  4500 0031 0000 4000 4011 b740 c0a8 0116  E..1..@.@..@....
        0x0010:  c0a8 0115 0e57 c025 001d ca62 0610 0420  .....W.%...b....
        0x0020:  0015 0401 3c00 2900 bce0 12ff 1f32 0100  ....<.)......2..
        0x0030:  80                                       .
07:13:17.274118 IP hassio.40680 > epv3le-gm.3671: UDP, length 10
        0x0000:  4500 0026 0755 4000 4011 aff6 c0a8 0115  E..&.U@.@.......
        0x0010:  c0a8 0116 9ee8 0e57 0012 839f 0610 0421  .......W.......!
        0x0020:  000a 0402 4100                           ....A.
07:13:17.276963 IP hassio.49189 > epv3le-gm.3671: UDP, length 10
        0x0000:  4500 0026 0756 4000 4011 aff5 c0a8 0115  E..&.V@.@.......
        0x0010:  c0a8 0116 c025 0e57 0012 839f 0610 0421  .....%.W.......!
        0x0020:  000a 0401 3c00                           ....<.
^[OA07:13:19.127297 IP epv3le-gm.3671 > hassio.40680: UDP, length 23
        0x0000:  4500 0033 0000 4000 4011 b73e c0a8 0116  E..3..@.@..>....
        0x0010:  c0a8 0115 0e57 9ee8 001f 7c22 0610 0420  .....W....|"....
        0x0020:  0017 0402 4200 2900 bce0 1273 1828 0300  ....B.)....s.(..
        0x0030:  800c 6f                                  ..o
07:13:19.127610 IP epv3le-gm.3671 > hassio.49189: UDP, length 23
        0x0000:  4500 0033 0000 4000 4011 b73e c0a8 0116  E..3..@.@..>....
        0x0010:  c0a8 0115 0e57 c025 001f 5fe6 0610 0420  .....W.%.._.....
[... 65 lines cut ...]
        0x0000:  4500 002a 4a34 4000 0211 75c5 c0a8 0115  E..*J4@...u.....
        0x0010:  e000 170c bf24 0e57 0016 b8f1 0610 0201  .....$.W........
        0x0020:  000e 0801 c0a8 0115 bf24                 .........$
07:13:20.587282 IP epv3le-gm.3671 > hassio.48932: UDP, length 78
        0x0000:  4500 006a 0000 4000 4011 b707 c0a8 0116  E..j..@.@.......
        0x0010:  c0a8 0115 0e57 bf24 0056 36ce 0610 0202  .....W.$.V6.....
        0x0020:  004e 0801 c0a8 0116 0e57 3601 0200 12ff  .N.......W6.....
        0x0030:  0000 01a0 017f 71f1 e000 170c 001b eb08  ......q.........
[... cut ...]
        0x0010:  c0a8 0116 9ee8 0e57 0018 83a5 0610 0207  .......W........
        0x0020:  0010 0200 0801 c0a8 0115 9ee8            ............
07:13:21.131892 IP epv3le-gm.3671 > hassio.40680: UDP, length 8
        0x0000:  4500 0024 0000 4000 4011 b74d c0a8 0116  E..$..@.@..M....
        0x0010:  c0a8 0115 0e57 9ee8 0010 c4f2 0610 0208  .....W..........
        0x0020:  0008 0200 c0a8 0116 0e57 3601 0200       .........W6...
^C
32 packets captured
32 packets received by filter
0 packets dropped by kernel`

The fourth try (after reboot), it also finds the gateway.

Julius2342, 2017-09-05:
hmmm, strange ... any ideas what could cause the difference?

TinusK, 2017-09-05:
Really not... Yesterday (and today for that matter) I wasn't at home... I use VPN and SSH to get into my Raspberry Pi (Hassbian). So, virtually no network traffic that could possibly delay a response...

jsl123, 2017-09-05:
Hi @TinusK, hi @Julius2342 

I'm just trying to reproduce this problem here.

My first thought about the multiple "tries" was: Is there more than one network interface involved?
Because I'd lot of duplicate datagrams on multicast sockets with multiple interfaces...
Are you sure you are using a python3 version >= 3.5?
Because with 3.4.x I also had *lots* of trouble with regard to multicast...

I'll post my findings when I get a connection here... :-)

jsl123, 2017-09-05:
OK. Here are some results from a test network setup over here:

```
10:13:33.742729 IP 172.31.120.23.51581 > 224.0.23.12.3671: UDP, length 14
        0x0000:  4500 002a 7c4c 4000 0211 e133 ac1f 7817  E..*|L@....3..x.
        0x0010:  e000 170c c97d 0e57 0016 1b6b 0610 0201  .....}.W...k....
        0x0020:  000e 0801 ac1f 7817 c97d                 ......x..}
10:13:33.743208 IP 172.31.120.45.3671 > 172.31.120.23.51581: UDP, length 78
        0x0000:  4500 006a 0000 4000 4011 f1ff ac1f 782d  E..j..@.@.....x-
        0x0010:  ac1f 7817 0e57 c97d 0056 304a 0610 0202  ..x..W.}.V0J....
        0x0020:  004e 0801 ac1f 782d 0e57 3601 0200 3400  .N....x-.W6...4.
        0x0030:  0000 01a0 017d b951 e000 170c 001b eb09  .....}.Q........
        0x0040:  9677 546f 705f 3131 3420 4b4e 584e 6574  .wTop_114.KNXNet
        0x0050:  2f49 5000 0000 0000 0000 0000 0000 0000  /IP.............
        0x0060:  0a02 0201 0301 0401 0501                 ..........
10:13:33.743231 IP 172.31.120.15.3671 > 172.31.120.23.51581: UDP, length 78
        0x0000:  4500 006a 0000 4000 4011 f21d ac1f 780f  E..j..@.@.....x.
        0x0010:  ac1f 7817 0e57 c97d 0056 34d7 0610 0202  ..x..W.}.V4.....
        0x0020:  004e 0801 ac1f 780f 0e57 3601 0200 4800  .N....x..W6...H.
        0x0030:  0000 01a0 0478 6c96 e000 170c 001b eb0a  .....xl.........
        0x0040:  2568 7461 6d61 735f 332e 352e 305f 7570  %htamas_3.5.0_up
        0x0050:  6461 7465 7465 7374 204b 4e58 4e65 7400  datetest.KNXNet.
        0x0060:  0a02 0201 0301 0401 0501                 ..........
10:13:33.743238 IP 172.31.120.126.3671 > 172.31.120.23.51581: UDP, length 78
        0x0000:  4500 006a fce2 4000 4011 f4cb ac1f 787e  E..j..@.@.....x~
        0x0010:  ac1f 7817 0e57 c97d 0056 293c 0610 0202  ..x..W.}.V)<....
        0x0020:  004e 0801 ac1f 787e 0e57 3601 0200 470c  .N....x~.W6...G.
        0x0030:  0000 01a0 1291 453a e000 170c b827 eb32  ......E:.....'.2
        0x0040:  6316 4170 704d 6f64 756c 6520 4b4e 584e  c.AppModule.KNXN
        0x0050:  6574 2f49 5000 0000 0000 0000 0000 0000  et/IP...........
        0x0060:  0a02 0201 0301 0401 0501                 ..........
```
Please note that there are three SEARCH_RESPONSes to (my) initial SEARCH_REQUEST send from `example_gatewayscannner.py`.

Command line and displayed result were as follows:
```
jsl@hawkwind:/xxxx/xknx$ /opt/bin/python3.6 examples/example_gatewayscanner.py 
Gateway found: Top_114 KNXNet/IP / 172.31.120.45:3671
- Device supports tunneling
- Device supports routing, connecting via 172.31.120.23
```
So obviously the "scanner" selects the first answer for the user. ;-)

HTH. If I should/can test anything for you, please comment

TinusK, 2017-09-05:
@jsl123 
Yes, I use Python 3.6:
```
pi@hassbian:~ $ python3
Python 3.6.0 (default, Sep  4 2017, 08:08:49)
[GCC 4.9.2] on linux
```
I ran the gatewayscanner on my raspberry pi which has only 1 ethernet port (removed the macs):
```
pi@hassbian:~ $ ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN mode DORMANT group default qlen 1000
```

jsl123, 2017-09-05:
@TinusK So there are at least two (relevant) interfaces: eth0 and wlan0 :-)
Both support mcast...

But I just had a "deeper look" onto the datagramms you send in your post from an hour ago. Most of these KNXnet/IP datagrams contain TUNNELING_REQUESTs and TUNNELING_ACKs ! These are "normal data" datagrams carrying cEMI KNX information...

Only the datagram at 07:13:20.586237 contains a search which is directly answered by the eibport in the following datagram. There's nothing special or faulty inside.

Julius2342, 2017-09-05:
@jsl123 : Thank you for helping to debug :-) I  appreciate any ideas for improvement.

I have to admit, the udp-abstraction was more trial and error, than a deep understanding of the internals of multicast and the correct setsockopts. The frustrating thing about this, that in the year 2017, we still have to deal with the C data structures of the late 70ies. 

The UDP/Multicast abstraction also always binds to a dedicated interface. The gateway scanner walks through all interfaces, sends a SearchRequest to each - and takes the first one as result. The idea behind this was, the most common use case is that you have one interface in your network. Do you think I should change this?

For security reasons, i also think it makes sense to bind multicast to an explicit interface. In my environment, i physically separated the KNX/IP network from the rest.

@TinusK : the timeout of the gateway scanner is set to 1 second. If the problem reoccurs, we might test increasing this timeout.

jsl123, 2017-09-05:
@TinusK Ohh. Last one first! ;-)
I think this one second timeout *is* the culprit, since the KNXnet/IP specification allows up to 4 seconds for responses (albeit for TUNNELING_ACKs)!
They had satellite links in mind... :-)

I think the binding to specific interfaces is very good (and needed). The implementation with sending on and listening to specific interfaces involved is also desireable.

And regarding the "C data structures" in 2017:
Please keep in mind that not every System in the building systems technology (or on the KNX bus) has got a Core i7 with 8GB RAM! ;-)
There are lots of KNX implementations which run on ARM M3 cores and even well below that down to old Motorola MC6809 controllers...

Thank you for effort into your KNX implementation! I love asynchronous implementations. :)

Julius2342, 2017-09-05:
Nothing is easier than increasing the timeout. Now XKNX is able to scan gateways which are located behind the moon!

And yes you are right :-) Somebody asked recently in the chat if XKNX could be used within embedded python for IoT devices speaking KNX/IP.

jsl123, 2017-09-05:
@TinusK But (as i wrote), there is "cEMI" traffic before your SERACH_REQUEST.
Can it be that there were multiple instances of the xknx stack running and (@Julius2342) they "fought" against each other?

Depending on how much (KNX) bus traffic must be handled and how much "jobs" are implemented in the eibport, it might be that it's KNX stack will not respond to the search request in one second...

TinusK, 2017-09-05:
@jsl123
You are right. Now that I think about it... I have the XKNX and KNX stack running simultaniously. But ETS kan connect and the xknx and knx stack in Hass work fine all together. I believe my eibPort3 can handle 3 simultanious connections.

TinusK, 2017-09-05:
In my eibPort3 documentation I see that I can set multiple. I will play with that setting and see if can reproduce the 'no gateway found' situation.

'''Individ. addr. for KNXnet/IP tunneling
These addresses are used by the tunneling connections established with the EIBPORT. One connection can 
be established for each address that has been specified. The addresses used here must not be similar to the 
"Physical address of the BCU", they have to be party addresses (they must not end with 0) and they must 
not be used by other devices in the KNX system. The first two digits of the addresses must correspond to 
the place of installation (or the used physical address). Since ETS 5 the ETS needs two parallel tunneling 
connections to successfully establish a connection. Therefore, at least two addresses have to be specified 
here. The addresses have to be separated by a semicolon.'''

http://bab-tec.de/index.php/eibport_v3_en.html

jsl123, 2017-09-05:
Yes. That's true. I think you can set up up to 8 (maybe more?) KNX tunneling addresses in the eibport. But keep in mind, that the ETS (at least Versions 4 and 5) need one connection for "internal" purposes...
That means if you have one ETS <-> eibport connection, you need at least two tunneling addresses.

"Party addresses"? :-) I think this should read "non-coupler" addresses.

Julius2342, 2017-09-06:
@jsl123 : Thank you for helping to debug this problem. 

May I ask you a question? Do you have any idea where to get the documentation for 

```
KNX HVAC Controller status.
Non-standardised DP type (in accordance with KNX AN 097/07 rev 3).
```

Many KNX manuals refer to this document, but i cant find it. I also contacted KNX.org, but they did not respond.

TinusK, 2017-09-06:
Can I help you in some way?
I have here several MDT Heating Aktuators... I control my heating using HVAC and would love to see and test this in HA using XKNX.

Julius2342, 2017-09-06:
That should work already... (The latest version from this morning also supports "controller status".)

Julius2342, 2017-09-06:
The only thing which is not working is if you want to set "setpoint shift" through KNX.

TinusK, 2017-09-06:
Ans thats exactly what I need :) I cant set a new temperature without using a setpoint offset.

TinusK, 2017-09-06:
But I will try to test HVAC modes and status today.

TinusK, 2017-09-09:
After I created more possible connections to the eibPort everything works perfektly stable!

Julius2342, 2017-09-09:
Great :-)

closed 2017-09-09 by Julius2342