XKNX/xknx #56: Sensor types Illuminance and speed_ms do not work correctly

TinusK, 2017-09-05:
My xknx sensor config in Home Assistant:
```
  - platform: xknx
    name: o_temperature
    type: 'temperature'
    address: '0/7/9'
  - platform: xknx
    name: o_wind
    type: 'speed_ms'
    address: '0/7/0'
  - platform: xknx
    name: o_brightness
    type: 'illuminance'
    address: '0/7/1'
```
Temperature reports 25.3 C and is correct.
Speed reports 0.0 m/s and is not correct. It stays also at 0.0 If I use the build-in 'KNX' (and not XKNX) then it returns correctly between 1.0 and 2.0 m/s.
Illuminance reports 0x3e,0x2a with no unit.

Julius2342, 2017-09-05:
for illuminance, may you just try "brightness" ?

TinusK, 2017-09-05:
Yes, 'brightness' works; it returns 2199.04 lx

TinusK, 2017-09-05:
BTW, HAss uses lux as unit for Brightness/Illuminance, not lx.

Julius2342, 2017-09-05:
Ok, i changed the naming from brightness to illuminannce. in version 0.7.11, probably you still run 0.7.10.

Regarding speed_ms, do you know the dpt-datatype? and may I ask you to change speed_ms to sam random value, to get the raw value?

Regarding lux/lx: I think lx is the SI unit?

TinusK, 2017-09-05:
Symbol is lx, name is Lux :)
[https://en.wikipedia.org/wiki/International_System_of_Units](url)

The wind speed (speed_ms) dpt is 9.005 speed (m/s) (= 2 Byte)
The raw data output is currently 0x00.0x1b

TinusK, 2017-09-05:
I see an ',' between the 0x00 and 0x1b... With Lux and Temp the digits are separated using a '.'

Julius2342, 2017-09-05:
0x00,0x1b is just the "raw" binary output.

Julius2342, 2017-09-05:
hmm, 0x00,0x1b should result in 0.27 ... would that be correct?

TinusK, 2017-09-05:
yes

Julius2342, 2017-09-05:
hmm, strange. May you add the following print to `home-assistant-plugin/custom_components/sensor/xknx.py` :

```
    @property
    def state(self):
        """Return the state of the sensor."""
        print(self.device)
        return self.device.resolve_state()
```

TinusK, 2017-09-05:
Ok, I put it in.
What should I do next?

Julius2342, 2017-09-05:
Run it again and watch for the output. Then we see if the problem occurs when reading the value or when displaying within HASS.

TinusK, 2017-09-05:
This you want?
```
<Sensor name="o_wind" group_address="<Address str="0/7/0" />" state="<DPTArray value="[0x0,0x0]" />" resolve_state="0x00,0x00" />
2017-09-05 17:14:22 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.o_wind, old_state=<state sensor.o_wind=unknown; friendly_name=Wind, icon=mdi:weather-windy @ 2017-09-05T19:14:14.739058+02:00>, new_state=<state sensor.o_wind=0x00,0x00; friendly_name=Wind, icon=mdi:weather-windy @ 2017-09-05T19:14:22.901247+02:00>>
2017-09-05 17:14:24 INFO (MainThread) [xknx.log] Ignoring group read for: <Address str="0/7/9" />
<Sensor name="o_temperature" group_address="<Address str="0/7/9" />" state="<DPTArray value="[0xc,0x56]" />" resolve_state="22.2" />
2017-09-05 17:14:24 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.o_temperature, old_state=<state sensor.o_temperature=unknown; unit_of_measurement=°C, friendly_name=Temperature @ 2017-09-05T19:14:14.734771+02:00>, new_state=<state sensor.o_temperature=22.2; unit_of_measurement=°C, friendly_name=Temperature @ 2017-09-05T19:14:24.458107+02:00>>
2017-09-05 17:14:24 INFO (MainThread) [xknx.log] Ignoring group read for: <Address str="0/7/0" />
2017-09-05 17:14:24 INFO (MainThread) [xknx.log] Ignoring group read for: <Address str="0/7/1" />
<Sensor name="o_brightness" group_address="<Address str="0/7/1" />" state="<DPTArray value="[0x34,0x55]" />" resolve_state="709.76" />
2017-09-05 17:14:24 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.o_brightness, old_state=<state sensor.o_brightness=unknown; unit_of_measurement=lx, friendly_name=Brightness, icon=mdi:brightness-5 @ 2017-09-05T19:14:14.743519+02:00>, new_state=<state sensor.o_brightness=709.76; unit_of_measurement=lx, friendly_name=Brightness, icon=mdi:brightness-5 @ 2017-09-05T19:14:24.592995+02:00>>
2017-09-05 17:14:24 INFO (MainThread) [homeassistant.components.http] Serving /api/camera_proxy/camera.weather_frankfurt to 192.168.1.5 (auth: True)
2017-09-05 17:14:24 INFO (MainThread) [xknx.log] Ignoring group read for: <Address str="0/7/2" />
<Sensor name="o_dusk" group_address="<Address str="0/7/2" />" state="<DPTArray value="[0x36,0x1a]" />" resolve_state="999.68" />
2017-09-05 17:14:24 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.o_dusk, old_state=<state sensor.o_dusk=unknown; unit_of_measurement=lx, friendly_name=Dusk, icon=mdi:weather-sunset @ 2017-09-05T19:14:14.747929+02:00>, new_state=<state sensor.o_dusk=999.68; unit_of_measurement=lx, friendly_name=Dusk, icon=mdi:weather-sunset @ 2017-09-05T19:14:24.672363+02:00>>
```

Julius2342, 2017-09-05:
yes. But it has the value 0x0,0x0 ... so we should wait for wind?

TinusK, 2017-09-05:
I'll be waiting with you :)

TinusK, 2017-09-05:
I'm sorry... somehow it works now. I think it was pure bad luck that the speed remained 0.0 for a very long time and after I changed the config it showed a value > 0. 

After starring at the 0.0 value for some minutes.... it showed 0.16 m/s for a second and returned to 0.0 (virtually no wind at home now). I confimed it with the eibPort visualization. It also showed briefly 0.16m/s.

I think you can close this issue.

TinusK, 2017-09-05:
Maybe you could tell me how to upgrade to 0.7.11?

Just do this again:
```
git clone https://github.com/XKNX/xknx.git
make install -C ~/xknx/home-assistant-plugin
```
and copy the new custom_components?

Julius2342, 2017-09-05:
a `git pull` within folder xknx should do it as well.

(The make install command just creates a symbolic link to the xknx directory)

Julius2342, 2017-09-05:
Im happy if problems disappear 👍

TinusK, 2017-09-05:
So I did a `sudo git pull` in the xknx folder:
```
pi@hassbian:~/xknx $ sudo git pull
remote: Counting objects: 9, done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 9 (delta 7), reused 9 (delta 7), pack-reused 0
Unpacking objects: 100% (9/9), done.
From https://github.com/XKNX/xknx
   d463df1..99bc78f  master     -> origin/master
Updating d463df1..99bc78f
Fast-forward
 test/sensor_test.py        | 13 +++++++++++++
 xknx/io/gateway_scanner.py |  2 +-
 2 files changed, 14 insertions(+), 1 deletion(-)
```
and copied the custom_components to the home assistant configuration folder.

In the custom_components/xknx.py it says:
`REQUIREMENTS = ['xknx==0.7.11']`

XKNX works in Home Assistant but if I change 'brightness' back to 'illuminance' then it still reports: 0x0c,0xb0

Did I upgrade wrong?

Julius2342, 2017-09-05:
hmm, 

The good news: upgrading of your local xknx checkout worked

the bad news: hass installs a second xknx library within `~/.homeassistant/deps/lib/python3.5/site-packages/xknx`. You can avoid this, by setting PYTHONPATH - but let me check if i can fix the package :)

TinusK, 2017-09-05:
I have:
`..deps/lib/python3.6/site-packages/xknx` and `..deps/lib/python3.6/site-packages/xknx-0.7.9-py3.6.egg-info`

Julius2342, 2017-09-05:
and did you manually "copy" the files to `~/.home-assistatnt/custom-components`. Make install should just create a symlink:

```
lrwxr-xr-x    1 julius  staff       81 Sep  5 20:28 custom_components -> /Users/julius/Desktop/homeautomation/xknx/home-assistant-plugin/custom_components
```

TinusK, 2017-09-05:
in `lib` there is only `python3.6`

TinusK, 2017-09-05:
yes, I manually copied the custom_components

Julius2342, 2017-09-05:
ok!

Julius2342, 2017-09-05:
May you please retry? I bumped version.

TinusK, 2017-09-05:
Got it.
Needed to remove deps and restart HA. Now it works.

TinusK, 2017-09-05:
Just for your info... I did not retry as you asked (read it to late).

Julius2342, 2017-09-05:
ah good 👍

No now everything is working?

TinusK, 2017-09-05:
Everything is working.

Julius2342, 2017-09-06:
Ok! Great! If any problems occur, please just open a ticket!

May I ask you to comment within original pull request in hass (8725) that everything is working?

closed 2017-09-06 by Julius2342