XKNX/xknx #57: Cover up/down and current position mixed up/wrong

TinusK, 2017-09-06:
My Cover config:
```
  - platform: xknx
    name: mf_toilet
    move_long_address: 2/0/60
    move_short_address: 2/1/60
    position_address: 2/2/60
    position_state_address: 2/3/60
    invert_position: true
```
The Up and Down buttons function correctly. Pressing Down (close_cover called), the covers move down. Pressing Up, the covers move up. Wenn cover is completely down however, the up arrow greys out and can't be pressed anymore. Same applies for when completely Up. Then the Down button greys out and cant be pressed anymore.

The Position slider is correct (0: up; 100: down). The mdi:icon continuously depicts open.

If I change the `invert_position` to false, then the buttons work as expected (cover completely down, only up arrow is active and vise versa) but the position slider says 0 wenn cover is completely down and 100 wenn cover is up which should be the other way around. The mdi:icon remains unchanged at `open` all the time.

Julius2342, 2017-09-06:
hmm, might be related with invert_position. May you try out if it works if you comment out this line?

(just everything should be inverted then)

(My idea is that the problem is related to the "is_closed" and "is_open" functionality ... and this has to be inverted too.

TinusK, 2017-09-06:
Commenting it out does not change. Results are the same as `invert_position: false`.

I'm looking thru the code and I'm wondering why `self.travelcalculator.start_travel(position)` (last line) is called even if it supports position?
```
    @asyncio.coroutine
    def set_position(self, position):
        """Move cover to a desginated postion."""
        if not self.supports_position:

            current_position = self.current_position()
            if position > current_position:
                yield from self.send(self.group_address_long, DPTBinary(1))
            elif position < current_position:
                yield from self.send(self.group_address_long, DPTBinary(0))
            self.travelcalculator.start_travel(position)
            return
        yield from self.send(self.group_address_position, DPTArray(position))
        self.travelcalculator.start_travel(position)
```

TinusK, 2017-09-06:
In `travelcalculator.py` one talks about time (seconds). But a cover position is expressed in percentage (0% is up; 100% is closed). Maybe there is a mix-up?

`self.send(self.group_address_position, DPTArray(position)`
`position` would here be in percentage.

TinusK, 2017-09-06:
The `position_address` and `position_state_address` are both expressed in percentage: DPT 5.001 percentage (0..100%)

Julius2342, 2017-09-06:
@TinusK : but internally this is calculated to a full byte, afaik?

(the travel calculator is started to get current positions inside hass while traveling. :-) )

TinusK, 2017-09-06:
What do you mean with internally? If I send 100 on the bus to the actuator the cover closes completely and returns the actual position on the bus as 100. 0 moves the cover to a complete open and returns 0.

Julius2342, 2017-09-08:
@TinusK : I pushed a fix, may I ask you to try out?

TinusK, 2017-09-08:
Can I do a simple git pull?
(remember... I use git for the first time and I'm not (yet) familiar with all the 'secret code words' like branch/master etc :)

TinusK, 2017-09-08:
With 'invert_position: true', the slide bar is correct, the icons and buttons are incorrect:
```
Sliding the bar to 100 closes the cover, 0 opens the cover => correct
The icon shows open when the cover is closed and closed when cover is actually open.
The 'Down' button opens the cover and 'Up' closes it.
```

With 'invert_position: false', the slide bar is incorrect, the icons are not working but buttons are correct:
```
Sliding the bar to 100 opens the cover, 0 closes the cover => incorrect
The icon shows open when the cover is closed and also open when cover is actually open => incorrect
The 'Down' button closes the cover and 'Up' opens it => correct
```

TinusK, 2017-09-11:
Could this be the reason why the icon always depicts open:

From xknx/devices/travelcalculator.py
```
        self.minimum_position_down = **256**  # excluding
        self.maximum_position_up = 0
```

From cover.py:
```
    def is_open(self):
        """Return if cover is open."""
        return self.travelcalculator.is_open()

    def is_closed(self):
        """Return if cover is closed."""
        return self.travelcalculator.is_closed()
```

XKNX/Cover return is_closed true when the travelcalculator maximum_position_closed equals current position.

The current position is 100 (due to 100%) and the maximum is 256. Ergo, dos not equal each other and is_closed = false;

travelcalculator.py:
```
    def is_open(self):
        """Return if cover is (fully) open."""
        return self.current_position() == self.maximum_position_up

    def is_closed(self):
        """Return if cover is (fully) closed."""
        return self.current_position() == self.minimum_position_down
```

The sliderbar in Home Assistant shows 0 at full left and 100 at full right. The KNX bus also returns the current position in %. I think the maximum should be set to 100 (iso 256) but I don't know what the reason was to set it to 256? Or am I completely missing something?

Julius2342, 2017-09-11:
hmm, internally KNX uses 0 - 255 as cover positions. This is calculated to 100-0 within hass plugin:

https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/cover/knx.py#L224

I suspect, that the problem is that the position is set to 255 and therefore does not match with the maximum position ...

TinusK, 2017-09-15:
config:
```
  - platform: xknx
    name: mf_dining
    move_long_address: 2/0/80
    move_short_address: 2/1/80
    position_address: 2/2/80
    position_state_address: 2/3/80
#    invert_position: true
```
Slider: 0(far left)=closed / 100(far right)=open    => incorrect (should be the other way round)
Buttons: Up=open / Down=close   => correct
Icon: remains unchanged either way (open or closed state) at open

config:
```
  - platform: xknx
    name: mf_dining
    move_long_address: 2/0/80
    move_short_address: 2/1/80
    position_address: 2/2/80
    position_state_address: 2/3/80
    invert_position: true
```
Slider: 0(far left)=open / 100(far right)=closed   => correct
Buttons: Up=close / Down=open   => incorrect (should be the other way round)
Icon: Open=closed / Closed=open    => incorrect (should be the other way round)

Conclusion:
For me (and someone else who has also the exact same problem) the 'invert_position=true' works the best (icons and slider are correct). Only the buttons should be the other way round.

Tinus

TinusK, 2017-09-15:
Updated previous comment with cover state info.

TinusK, 2017-09-15:
May I ask why we need an invert switch for KNX Covers? If something is wrong (usually someone switched the wiring at the motor) it can be corrected in the actor itself or using ETS (group addresses)...

Julius2342, 2017-09-15:
well, i don't know. It was just implemented in the previous implementation ... so i added it :)

TinusK, 2017-09-15:
if I comment out the 'if not invert'
```
    @staticmethod
    def from_knx_position(raw, invert):
        """Convert KNX position [0...255] to hass position [100...0]."""
        position = round((raw/256)*100)
        # if not invert:
        #     position = 100 - position
        return position

    @staticmethod
    def to_knx_position(value, invert):
        """Convert hass position [100...0] to KNX position [0...255]."""
        knx_position = round(value/100*255.4)
        # if not invert:
        #     knx_position = 255-knx_position
        print(value, " -> ", knx_position)
        return knx_position
```
the slider works correct with 'invert_position=false', the buttons work correct but the down button is greyed out when the cover is fully open and vice versa.

TinusK, 2017-09-15:
```
    @staticmethod
    def from_knx_position(raw, invert):
        """Convert KNX position [0...256] to hass position [100...0]."""
        position = round((raw/256)*100)
        if not invert:
            position = 100 - position
        return position

    @staticmethod
    def to_knx_position(value, invert):
        """Convert hass position [100...0] to KNX position [0...256]."""
        knx_position = round((value/100)*256)
        if not invert:
            knx_position = 256 - knx_position
        print(value, " -> ", knx_position)
        return knx_position
```
I've changed the 255.4 -> 256 and put brackets around value/100.

With 'invert_position=false' the icons work almost. If I press down, the covers go down. After they are completely closed, the icon changes to 'closed'. A few seconds later, it changes back to open. Most probably the KNX bus reports back the current position: 100 and something gets messed up.

Julius2342, 2017-09-17:
@TinusK : May I ask you to try out?

Julius2342, 2017-09-17:
@twistermcneal , man with the exotic cover actors without direct positioning, may I ask you to test if still everything works as expected on your setup? (I think your use case is covered with a unit test, but I'm not sure).

TinusK, 2017-09-17:
@julius2342: I will as soon as I'm backbhome (day after tomorrow). I see you have removed the dpt5 and changed it to dpt5001 (0-100). Perfekt! Looks good and I'm eager to test it!

TinusK, 2017-09-18:
@Julius2342 
[xknx-cover.zip](https://github.com/XKNX/xknx/files/1310061/xknx-cover.zip)
My brother has pretty much the same setup as I have and has exactly the same problem with the xknx cover in HAss. The video above is from him. He did a git pull and copied the new custom_components to the config.

As you can see; using invert_position=true, the buttons, icon and state are correct. The slider does not move and jumps to 0 when the cover is fully closed and jumps to 100 when the cover is fully open.

The slider itself works. Putting the slider on 80 sets the cover to 20%; setting it to 0 sets the cover to 100%.

Does this help you further?

Julius2342, 2017-09-18:
@TinusK : so the problem still occurs with the new version from yesterday evening? (0.7.14)

I did a complete refactoring of the cover component - it now works on my setup, also with inverted cover. I also fixed the problem that HASS still reported an open cover after having it closed.

TinusK, 2017-09-18:
Please Disgerard previous post... He got something mixed up... I will test it myself and report back to you.

TinusK, 2017-09-18:
Ok, I've tested it. I'm not at home but checked it against my eibPort visu.
```
  - platform: xknx
    name: mf_toilet
    move_long_address: 2/0/60
    move_short_address: 2/1/60
    position_address: 2/2/60
    position_state_address: 2/3/60
    invert_position: false
```
Everything works except the slider is wrong. Putting the slider on 80 sets the cover to 20%; setting it to 0 sets the cover to 100% and vise versa which is counter intuitive.

TinusK, 2017-09-18:
In other words:
`invert_position: false`
buttons: correct
icon: correct
state: correct
slider: works but is inverted (it does not move during movement anymore which is fine by me).

Julius2342, 2017-09-18:
Ideally, we should do this analog to other cover components. I also do not know if HASS internally sets the position to "0" or "100" if the cover is closed ...

This has a potential risk of messing it up :)

TinusK, 2017-09-19:
Yes, you are right. But it is not intuitive to use the slider with covers :(

But I must say, excellent job on reprogramming the xknx cover!

Could we now tackel HVAC Setpoint shift? :)
https://github.com/XKNX/xknx/issues/48
Its for me the only thing missing...

Julius2342, 2017-09-19:
Yes, that is next on my list! (And i need it, bc here in Bavaria, the winter is coming closer).

So we can close this issue?

TinusK, 2017-09-19:
Yes, we can close it!
Although I live in Hessen, I just came back from Munich and yes, it's cold over there :)

closed 2017-09-19 by TinusK