XKNX/xknx #58: License of components not clear

bachp, 2017-09-09:
The current license of the component is not consistent.

The repo includes the LGPL-3.0 license while the PyPi package sais it is MIT and at the same time LGPLv3.

If non of the contributors (@Julius2342, @twistermcneal, @andreasnanko, @cyberjunky) have any objections it I suggest putting the package under MIT as the license allows more flexibility. If you agree I can create a PR doing this.

If putting the component under MIT is not desired please correct the entry in https://github.com/XKNX/xknx/blob/master/setup.py#L18

Julius2342, 2017-09-09:
Valid point!

I tend to say, that MIT is a bit too loose - but on the other hand, I'm happy if as many people as possible use xknx.

We could also consider changing to Apache-License as [home-assistant](https://github.com/home-assistant/home-assistant/blob/dev/LICENSE.md), [aiohttp](https://github.com/aio-libs/aiohttp/blob/master/LICENSE.txt) do..

(My personal favorite would be [Beerware](https://en.wikipedia.org/wiki/Beerware) , just kidding :-) )

Julius2342, 2017-09-09:
(the more i think of it, i think lgpl is definitely not the best choice - lgpl had more the shipping of binary .so in mind ...)

bachp, 2017-09-09:
If you choose to use Apache-2.0 is suggest to have it dual licensed as is is incompatible with some versions of GPL.

But from my experience MIT is just the simplest.

Beerware is nice too :)

bachp, 2017-09-09:
Oh one more thing. I don't know how much code comes from knxip which is MIT.

PR #59 (2017-09-10): Reclicense under MIT
Closes #58
closed 2017-09-10 by Julius2342