XKNX/xknx #60: loop.create_datagram_endpoint does not work with Windows

Julius2342, 2017-09-18:
Problem description
--------------------

Under Windows you get the following error when connecting to KNX/IP device:

```bash
2017-09-17 18:35:45 ERROR (MainThread) [homeassistant.setup] Error during setup of component knx
Traceback (most recent call last):
  File "D:\Python36\lib\site-packages\homeassistant\setup.py", line 191, in _async_setup_component
    result = yield from component.async_setup(hass, processed_config)
  File "D:\Python36\lib\site-packages\homeassistant\components\knx.py", line 78, in async_setup
    yield from hass.data[DATA_KNX].start()
  File "D:\Python36\lib\site-packages\homeassistant\components\knx.py", line 138, in start
    connection_config=connection_config)
  File "D:\HomeAssistant\deps\Python36\site-packages\xknx\xknx.py", line 66, in start
    yield from self.knxip_interface.start()
  File "D:\HomeAssistant\deps\Python36\site-packages\xknx\io\knxip_interface.py", line 76, in start
    self.connection_config.gateway_port)
  File "D:\HomeAssistant\deps\Python36\site-packages\xknx\io\knxip_interface.py", line 106, in start_tunnelling
    yield from self.interface.start()
  File "D:\HomeAssistant\deps\Python36\site-packages\xknx\io\tunnel.py", line 73, in start
    yield from self.connect_udp()
  File "D:\HomeAssistant\deps\Python36\site-packages\xknx\io\tunnel.py", line 79, in connect_udp
    yield from self.udp_client.connect()
  File "D:\HomeAssistant\deps\Python36\site-packages\xknx\io\udp_client.py", line 167, in connect
    remote_addr=self.remote_addr)
  File "D:\Python36\lib\asyncio\base_events.py", line 934, in create_datagram_endpoint
    raise exceptions[0]
  File "D:\Python36\lib\asyncio\base_events.py", line 921, in create_datagram_endpoint
    yield from self.sock_connect(sock, remote_address)
  File "D:\Python36\lib\asyncio\proactor_events.py", line 451, in sock_connect
    return self._proactor.connect(sock, address)
  File "D:\Python36\lib\asyncio\windows_events.py", line 511, in connect
    ov.ConnectEx(conn.fileno(), address)
OSError: [WinError 10022] Argomento fornito non valido [NOTE TO READER -- IN ENGLISH: Supplied argument is invalid]
```

Operating system: Windows
Python version: 3.6

The corresponding code fragment (`udp_client.py`)  is:

```
(transport, _) = yield from self.xknx.loop.create_datagram_endpoint(
                lambda: udp_client_factory,
                local_addr=self.local_addr,
                remote_addr=self.remote_addr)
```

If you google for the error you find similar problems: https://github.com/chrysn/aiocoap/issues/27 

Help needed!
-------------

I have no experience with Windows and no idea how to fix this. Help needed!

encbladexp, 2017-11-08:
[Link to Python Doc](https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.AbstractEventLoop.create_datagram_endpoint) says, the ProActorEventLoop doesn't provide the required stuff at all. Maybe we must simply choose an different Event Loop?

farmio, 2021-01-10:
There have been changes to support Windows a while ago. Think this should be solved.

closed 2021-01-10 by farmio