XKNX/xknx #61: Cover: dedicated stop address

Julius2342, 2017-09-18:
Problem description
---------------------

At the moment covers are stopped via the short up/down step address. This works for Gira and some over cover actors.

Some cover actors have a dedicated stop address, e.g. http://www.mdt.de/download/MDT_THB_Jalousieaktor_02.pdf  (page 12)

The datatype is 1.001 – 1 Bit.

This would also make the home - assistant configuration less confusing.

Current workaround
---------------------

The current workaround is to define a short up/down address:

```
- platform: knx
  name: "Livingroom.Cover2"
  move_long_address: '3/0/16'
  move_short_address: '3/0/17'
```

encbladexp, 2017-11-08:
Using long/short addresses for covers is the knx standard. In most common installations each cover has three communication objects, and therefor groups connected:
 * short (up/down, short rotations and stop)
 * long (up/down, long run)
 * alarms (at least wind is common, sometimes dedicated objects for wind and rain or 1-3 free configureable communication objects)

I never seen an, used, dedicated stop object in production.

Julius2342, 2017-11-09:
Thank you for your input, lets close the issue then :)

closed 2017-11-09 by Julius2342