XKNX/xknx #63: KNX scenes

fdandersson, 2017-09-18:
I would like the possibility to trigger KNX scenes.

The switch/button/function would need to be able to send value from 0 to 63 (1-64) in 1 Byte form, to a specified address.

Like: Triggering a value “23” to 1/1/23 will trigger scene “23” which could be “all lights on on ground floor”

**Datatypes:**

> KNX/EIB Function type: Scene | Information length: 1 Byte | DPT type: DPT 17 | Value: 0...63
> 
> Source: https://www.domotiga.nl/projects/selfbus-knx-eib/wiki/Datatypes
>

vandijks, 2017-10-05:
+1 Would love to see scenes supported too.

Julius2342, 2017-10-05:
will do after i have finished with climate support ...

iditude, 2018-01-16:
There is also another use case that involves sending value from Home Assistant to a KNX address. For instance I can control my heater by sending a specific value to an address (0 for normal operation, 4 for antifreeze operation). I thought this was the purpose of the notify platform but it doesn't do the trick.

Hence I would consider the possibility of sending values to addresses as the generic ask (not just scenes).

Thanks

Julius2342, 2018-01-16:
@iditude : https://github.com/XKNX/xknx/pull/93 is a first step in this direction. Creating virtual sensors which expose a local value to KNX bus. This concept might be extended to other values.

But as always, things take longer then expected.

Julius2342, 2018-01-16:
did you try to control this via a climate control? They support these climate-operation modes ... you could also instanciate a climate component and hide it from interface.

iditude, 2018-01-17:
@Julius2342: This solved my issue. Thanks for suggesting the climate component, I had no idea operation_mode was part of this. Awesome. Man you're really sharp on the whole KNX thing. Bravo!

Julius2342, 2018-01-21:
Yay, finally, basic scene support. May somebody please try out? I dont have scenes at home ...

closed 2018-02-04 by Julius2342