XKNX/xknx #64: Venetian Blinds - Angle Address and State not working

Karssi, 2017-09-21:
Hi.

After the recent update for inverted control and slider for covers I noticed that the Controls in HAss for my Venetian Blinds (Raffstores) were missing (without changes to the .yamls).
After some trouble shooting I figured out that when the position_address and position_state_address were commented out, the controls were back. But I can now only control the up and down movement of the blinds. I no longer have direct control of the angle (the second slider is also missing when you open the cover controls in HAss).

current cover configuration (working, but up/down only)
  - platform: xknx
    name: mf_dining_e
    move_long_address: 3/0/1
    move_short_address: 3/1/1
    position_address: 3/2/1
    position_state_address: 3/3/1
#    angle_address: 3/4/1
#    angle_state_address: 3/5/1
    travelling_time_down: 81
    travelling_time_up: 83
    invert_position: false
    invert_angle: false

old cover configuration (used to work up/down and with direct angle control)
  - platform: xknx
    name: mf_dining_e
    move_long_address: 3/0/1
    move_short_address: 3/1/1
    position_address: 3/2/1
    position_state_address: 3/3/1
    angle_address: 3/4/1
    angle_state_address: 3/5/1
    travelling_time_down: 81
    travelling_time_up: 83
    invert_position: false
    invert_angle: false

Karssi

Julius2342, 2017-09-21:
Which version do you have running? (0.53.1 or current dev?)

TinusK, 2017-09-21:
@julius2342
I know him... We talked about the cover problems I had... He had exactly the same problems :)
He did (as I did) a git pull and updated the custom_components. So he should have version 0.7.14. I don't have raf stores so I can't test it.

TinusK, 2017-09-21:
But, if I add the lines:
```
    angle_address: 2/2/90
    angle_state_address: 2/3/90
```
to my present working config (knx addresses are fake/same as position addresses)... the entire cover is not visible in hass anymore. If I comment those two lines out (remove them) then the cover is visible again.

Maybe this will help.

Julius2342, 2017-09-21:
I don't have venetian blinds, so i did not try it out %)

Will try to fix before new version gets released. Should be some trivial error.

Julius2342, 2017-09-21:
I think this is the problem:

```
diff --git a/home-assistant-plugin/custom_components/cover/xknx.py b/home-assistant-plugin/custom_components/cover/xknx.py
index 31562e3..e7b0792 100644
--- a/home-assistant-plugin/custom_components/cover/xknx.py
+++ b/home-assistant-plugin/custom_components/cover/xknx.py
@@ -181,7 +181,7 @@ class KNXCover(CoverDevice):
         """Return current tilt position of cover."""
         if not self.device.supports_angle:
             return None
-        return self.device.get_angle()
+        return self.device.current_angle()
```

will commit later today

Julius2342, 2017-09-21:
@Karssi : May you please test?

Karssi, 2017-09-21:
Hi Julius. I am not at home today. Will test first thing tomorrow morning.

Karssi, 2017-09-22:
Hi.

With the new update it is working again.

Julius2342, 2017-09-22:
Yay :-)

closed 2017-09-22 by Julius2342