XKNX/xknx #65: Make state_updater adjustable by config file (On/OFF)

twistermcneal, 2017-09-21:
https://github.com/XKNX/xknx/blob/b1da2bb925d6eed4a96ffa6157e67063ed4a047c/home-assistant-plugin/custom_components/xknx.py#L137

yield from self.xknx.start(
            state_updater=True,
            connection_config=connection_config)

Julius2342, 2017-09-21:
May you try out?

Karssi, 2017-09-22:
Hi

If I change:
        yield from self.xknx.start(
            state_updater=self.config[DOMAIN][CONF_XKNX_STATE_UPDATER],
            connection_config=connection_config)

to:
        yield from self.xknx.start(
            state_updater=True,
            connection_config=connection_config)

in custom_components\xknx.py then nothing is working anymore. All KNX elements are then gone.
I don't know if this is the change you wanted me to try out. I am new to Github, HAss and its underlying OS.

Julius2342, 2017-09-22:
Uh thats not good :-/ Will investigate later ...

twistermcneal, 2017-09-29:
works perfectly for me, nice one
sry for late testing, i`am really busy at the moment....

closed 2017-09-29 by twistermcneal

rsterrenburg, 2017-09-30:
Thanks for adding this!