XKNX/xknx #67: KNX Humidity sensor displays data as hexdecimal number.

Karssi, 2017-09-23:
Hi.

I have a KNX Temp/Humidity sensor installed in my bedroom. Wenn I add the sensor to HAss via the following config:
  - platform: xknx
    name: lf_masterbedroom_humidity
    type: percent
    address: 5/1/0
I get a readout in HAss where the number is in hexdecimal (e.g. 0x0e,0x73) in stead of in a decimal format.
If I change type: percent to type: temperature I get the correct value in decimal but then I have the C suffix behind the number.

The DPT in KNX is 9.007 (humidity %), 2-byte

Julius2342, 2017-09-25:
Hi @Karssi ,

I added that functionality to `master`. May I ask you to test? 

You have to change the config to `type: humidity`.

Karssi, 2017-09-25:
Hi.

unfortunately it didn't change anything. changed type to humidity. I still get the data in hexdecimal but now without a % suffix.

Julius2342, 2017-09-25:
how did you test it? (just want to make sure you are running the current version)

Karssi, 2017-09-25:
Ok..

logged in via ssh, then in the xknx directory did a sudo git pull. It then downloaded the new files.
Stopped Hass, in the home assistent folder deleted the custom_components en deps folders.
copied the custom_components folder from the xknx folder to the home assistant folder.
Then restarted HAss.

Karssi, 2017-09-25:
xknx

Julius2342, 2017-09-25:
may you delete the line `REQUIREMENTS` from the file: `custom_components/xknx.py`

then please do a `rm -rf ~/.homeassistant/deps/lib/python/site-packages/xknx*`

and please also make sure that PYTHONPATH set set to xknx directory.

Karssi, 2017-09-25:
Hi Julius.

Please keep in mind I am a Windows man and completely new to 'linux'. sorry..
I deleted the REQUIREMENTS from the custom_compontents/xknx.py
I then did the rm command.

How do I check/set PYTHONPATH? 
(because HAss is not working anymore at the moment. I get a connection refused message)

I am using the HASSBian image and manually updated Python to v3.6

Julius2342, 2017-09-25:
ok :)

you can start hass by directly typing `hass` on command line.

PYTHONPATH is an environment variable. You can set it via `export PYTHONPATH=/path/to/xknx/`. This environment is then valid for this terminal session. (so when you type in hass in the same window, it takes the this variable.

Can explain further later ...

Karssi, 2017-09-25:
Thanks. :-)

Did all that.
But the output is still in hexadecimal.

Julius2342, 2017-09-25:
hmm .. my feeling is, that somehow still the old version is running. 

If you look at https://github.com/XKNX/xknx/blob/master/xknx/devices/sensor.py#L81 you can see, that the logic for determination of the correct unit is no real magic. If the value is just displayed in raw format, you know that it did not find the right `elif` statement.

You could add a `print(self.value_type)` to see, if a) you are in the right lib :-) b) value_type is set correctly to `humidity` and trace down why it does choose the correct unit / value calculation.

Karssi, 2017-09-28:
Hi Julius.

Sorry for the late answer. Didn't have the time to look at it..
I still couldn't get it to work with just the type: humidity.
I found a work around with the config:
  - platform: xknx
    name: lf_masterbedroom_humidity
    type: temperature
    address: 8/1/90
    unit_of_measurement: %

I will revisit it when there is new HASSbian image with Python 3.6 and xknx so that I can reinstall from scratch.

Thanks for your time..

Karssi, 2017-11-01:
Hi Julius,

After new HASSBIAN Install. It works as advertised! :-)

Julius2342, 2017-11-01:
Great :-) I will close the issue then!

closed 2017-11-01 by Julius2342