XKNX/xknx #68: Support for generic KNX Datapoints

flo-wer, 2017-09-25:
Hello,

it seems this KNX library doesn't support setting arbitrary datapoints for each sensor but instead uses a fixed mapping between the sensor type and the datapoint. As the KNX standard defines all possible data types, it would be nice to support using just the plain datapoint as type:
```
  - platform: knx
    name: Kitchen.Temperature
    address: '6/2/1'
    type: '9.001'
```

Currently I have sensors with the following types that aren't supported:
7.013 DPT_Brightness
9.006 DPT_Value_Pres
9.008 DPT_Value_AirQuality
9.028 DPT_Value_Wsp_kmh
14.033 DPT_Value_Frequency
14.056 DPT_Value_Power

Also the current defined types aren't unique, e.g. both 9.005 DPT_Value_Wsp and 14.065 DPT_Value_Speed use m/s as unit.

iditude, 2017-12-11:
+1 on this. I use KNX to monitor my power consumption and it uses a data type 13.010 active power (in W/h)

This would be super helpful!

encbladexp, 2017-12-11:
Sound like a good idea for me too.

JohanElmis, 2018-01-07:
For sensors I think we have this covered in a Development fork. In for review and merge with upstream development in the near future.

Julius2342, 2018-01-21:
Is there anything missing here?

flo-wer, 2018-01-21:
Mhh... after looking at the patch from @JohanElmis, I'm not really convinced how this changes the way the sensor type is defined. After all, only additional Data Points were added, but I thought of a way to have all Data Points covered.
From looking at the code, I think that using a new class for each Data Point doesn't scale very well. Wouldn't it be better to just use a list for all Data points?

Something I quickly hacked together:
```
class DPT2ByteFloat(DPTBase):

	def __init__(self, value_min = -671088.64, value_max = 670760.96, unit = '', resolution = 1, payload_length = 2):
		self.value_min = value_min
		self.value_max = value_max
		self.unit = unit
		self.resolution = resolution
		self.payload_length = payload_length

DPT_MAP = {
	'9.xxx': DPT2ByteFloat(None,	None,	''),	# generic DPT2ByteFloat
	'9.001': DPT2ByteFloat(-273,	670760,	'°C'),	# DPT_Value_Temp
	'9.002': DPT2ByteFloat(-670760,	670760,	'K'),	# DPT_Value_Tempd
	'9.003': DPT2ByteFloat(-670760,	670760,	'K/h'),	# DPT_Value_Tempa
	'9.004': DPT2ByteFloat(0,	670760,	'Lux'),	# DPT_Value_Lux
	'9.005': DPT2ByteFloat(0,	670760,	'm/s'),	# DPT_Value_Wsp
	'9.006': DPT2ByteFloat(0,	670760,	'Pa'),	# DPT_Value_Pres
	'9.007': DPT2ByteFloat(0,	670760,	'%'),	# DPT_Value_Humidity
	'9.008': DPT2ByteFloat(0,	670760,	'ppm'),	# DPT_Value_AirQuality
	'9.010': DPT2ByteFloat(-670760,	670760,	's'),	# DPT_Value_Time1
	'9.011': DPT2ByteFloat(-670760,	670760,	'ms'),	# DPT_Value_Time2
	'9.020': DPT2ByteFloat(-670760,	670760,	'mV'),	# DPT_Value_Volt
	'9.021': DPT2ByteFloat(-670760,	670760,	'mA'),	# DPT_Value_Curr
	'9.022': DPT2ByteFloat(-670760,	670760,	'W/m²'),# DPT_PowerDensity
	'9.023': DPT2ByteFloat(-670760,	670760,	'K/%'),	# DPT_KelvinPerPercent
	'9.024': DPT2ByteFloat(-670760,	670760,	'kW'),	# DPT_Power
	'9.025': DPT2ByteFloat(-670760,	670760,	'l/h'),	# DPT_Value_Volume_Flow
	'9.026': DPT2ByteFloat(None,	None,	'l/m²'),# DPT_Rain_Amount
	'9.027': DPT2ByteFloat(-459.6,	None,	'°F'),	# DPT_Value_Temp_F
	'9.028': DPT2ByteFloat(0,	None,	'Lux'),	# DPT_Value_Wsp_kmh

	'14.xxx': DPT4ByteFloat(None,	None,	''),		# generic DPT4ByteFloat
	'14.000': DPT4ByteFloat(None,	None,	'ms⁻²'),	# DPT_Value_Acceleration
	'14.001': DPT4ByteFloat(None,	None,	'rad s⁻²'),	# DPT_Value_Acceleration_Angular
	# ...
}

DPT_ALIAS = {
	'temperature':	'9.001',
	'humidity':	'9.007',

	# ...

	'DPT-9':	'9.xxx',
	'2 byte float':	'9.xxx',

	# ...

	'DPT-14':	'14.xxx',
	'4 byte float':	'14.xxx',
}
```

First lookup in DPT_ALIAS, then in DPT_MAP.

farmio, 2020-08-18:
Most DPT are implemented now. They are configurable by number or string (alias). Generic types (eg. `9`) are also configurable.

closed 2020-08-18 by farmio