XKNX/xknx #69: Error thrown when Reading DPT 9.004

ssoerensen, 2017-09-25:
I have setup a sensor in home assitant. and when ever the lux value updates a error is thrown, however the value is update.

ETS Group
 | 3/2/0 | Driveway-SEN-LUX |   | No | No | lux (Lux) | 2 bytes |   |  

Error message:
2017-09-25 11:42:17 ERROR (MainThread) [xknx.log] Exception while processing telegram: unsupported operand type(s) for &: 'tuple' and 'int'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/xknx/core/telegram_queue.py", line 99, in process_telegram
    yield from self.process_telegram_incoming(telegram)
  File "/usr/local/lib/python3.6/site-packages/xknx/core/telegram_queue.py", line 130, in process_telegram_incoming
    yield from device.process(telegram)
  File "/usr/local/lib/python3.6/site-packages/xknx/devices/binary_sensor.py", line 133, in process
    binary_value = telegram.payload.value & bit_masq != 0
TypeError: unsupported operand type(s) for &: 'tuple' and 'int'

Julius2342, 2017-09-25:
You have to define this device as a "sensor" not a "binarysensor". Then it should work.

ssoerensen, 2017-09-25:
It is defined as a sensor
`sensor:
  - platform: knx
    name: Driveway-SEN-LUX
    type: illuminance
    icon: mdi:theme-light-dark
    address: '3/2/0' # Lux over 300 turn of lights
`

Julius2342, 2017-09-25:
The stacktrace just happens somewhere within binary_sensor. So it looks like you have defined a binary_sensor with a group_address which is receiving a two-byte KNX telegram.

ssoerensen, 2017-09-25:
Oh fuck. i'm just going to close this one again.. i have same group address referenced as a sensor as well *FACE PALM*!

closed 2017-09-25 by ssoerensen

Julius2342, 2017-09-25:
happens :-)

But we should make the error message more readable ....