XKNX/xknx #70: KNX Cover State not correct in HASS

FireFrei, 2017-10-03:
When restarting HASS (0.54.0) my KNX covers (actors are made by ABB) show an incorrect state - actually they are open, but hass says closed. 
I verified group addresses, I think they are correct. Object is marked as readable in ETS.

The abb documentation says:

```
 "Status Höhe/Lamelle 0…255" 1 Byte 
Der Ausgang sendet die relative Position des Behangs und die Lamellenstellung auf zwei getrennten Kommunikationsobjekten jeweils als 1-Byte-Wert (0…255). Für die Position des Behangs gilt: Der Wert 0 entspricht der Position oben (0 %). Der Wert 255 entspricht der Position unten (100 %).  
```

I tried to set the "invert_position" flag to "true" in hass, but:
- On hass startup, cover is still shown as closed but is actually opened
- Movement of covers is now inverted (Button "down" -> cover goes up)

I did not notice any knx errors regarding xknx covers in the hass logs.
When I move the knx cover manually (by a native knx switch/button) the state is updated correctly in hass, after the cover has stopped.

Julius2342, 2017-10-03:
*sigh* working on it. Looks like i messed up the cover abstraction.

Julius2342, 2017-10-03:
did not want to close it ...

Julius2342, 2017-10-03:
I pushed a new version, may I ask you to have a look?

FireFrei, 2017-10-07:
Well, you need to help me a bit. how can I get the new release working without using the custom_component in hass?

I installed the new version of xknx, this results in some errors starting hass - I guess I need to update the hass component as well. but I'm not using it via the custom_component, I use the hass "native" component. Where can I find the code?

Julius2342, 2017-10-07:
when not using custom component you have to install a development environment. you have to clone https://github.com/XKNX/home-assistant.git and follow the steps for setting up a development environment.

May you join https://discord.gg/5XARFNT then we can help more interactively ..

Julius2342, 2017-10-07:
(a wrong notification group address blocked the state updater.)

commit 254ea211 (2017-10-03): Should fix: #70 https://github.com/home-assistant/home-assistant/issues/9414
closed 2017-10-07 by Julius2342