XKNX/xknx #71: Question: Should all lights turn on on startup

mariwing, 2017-10-04:
Hi guys, great that you are working on improving knx support in Home Assistant. With both the old (pre-xknx as I understand it) knx implementation and the new one all my lights turn on in the house when Home Assistant is starting up. Is this the expected/normal behavior? Because it is a little bit inconvenient ;-)

Julius2342, 2017-10-04:
> Uh, definitely not expected behavior. I suspect the state-updater going
wild.

The state updater sends a group read for all devices on startup - which
should not change the state ...

The next version (probably shopped with the Hass version after the next
version) will ship with a configuration option to turn that off.

You may use the telegram monitor which ships with xknx to see which
telegrams are causing this behaviour.
-- 
Sent from mobile

Julius2342, 2017-10-05:
Beyond disabeling the state updater I would like to understand the underlaying problem.

What happens if you send a group read to the group address of your light? May I ask you to try out with the following example: https://github.com/XKNX/xknx/blob/master/examples/example_value_reader.py . You have to modify the internal group address in line 13.

twistermcneal, 2017-10-05:
while i´am sitting in the airplan back to germany(finally) i read your post and i think the problem is knx specific ... if you have a central knx function that turns all lights on/off and you send a group read to this address then all actuators will read the central state and switch too that state ... hope that helps

Julius2342, 2017-10-05:
the daily KNX-WTF :)

twistermcneal, 2017-10-05:
The word engineer comes from a Latin word meaning "cleverness" ... I think knx engineers are the smartest :)
when you think about the security aspect of this problem ,... horrifying
but security and knx are two different worlds...

mariwing, 2017-10-05:
Thanks for these witty remarks, made my day :-) I do have a group address for all lights, it is connected to a switch by my entrance door so I press it whenever I leave the house. This address has not been specifically added to Home Assistant, but  if there is some kind of "full scan" going on at startup this could very well be the explanation. Also; I never push "all lights" on since that is not needed so the state of this group address is most likely always off. 

Unfortunately I don't know anything about running the python script that was linked. I have a separate PI setup as the KNX USB-TO-IP device so I could probbably try to run the script on that machine. I will give it a try, just have to google "run python script on raspberry pi" first :-)

Julius2342, 2017-10-05:
Installing dependencies:

```bash
sudo pip3 install PyYAML
sudo pip3 install netifaces
```

Getting xknx (not via pip, bc we want to edit/use an example:

```bash
git clone https://github.com/XKNX/xknx.git
cd xknx
```
Edit `examples/example_value_reader.py` with your favorite editor. Line 13 contains the group address.

Run example from xknx directory:

```bash
export PYTHONPATH=`pwd`
python3 examples/example_value_reader.py
```

The result should be similar to:

````
<Telegram group_address="<Address str="6/2/1" />", payload="<DPTArray value="[0xc,0x51]" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />
````

(there might also be an error afterwards but that can be ignored)

mariwing, 2017-10-05:
Thanks man - you saved me a lot of googling. But you forgot I am a Python noob so I had to do 

`sudo apt-get install python3-pip` 

before the rest and then figuring out that I needed Python 3.5 which I installed from here: https://gist.github.com/elBradford/5e0c999b6d2ab8fcf5331c5ac177a486

That asside; the output of the command sent to one of my lights became:

  File "examples/example_value_reader.py", line 8
    async def main():
            ^
SyntaxError: invalid syntax

So sorry - no luck! I have ETS 3 and I did restart Home Assistant with ETS 3 busmonitor running. There was a "storm" of both read and writes for many of the reads issued from Home Assistant/xknx

Julius2342, 2017-10-05:
I wonder, how could hass work without having python3 installed?

The error message sounds like you are still using an outdated python version.

TinusK, 2017-10-05:
As far as I can tell, no commands are written to the bus by the state updater right? It only asked the states for the given state addresses...

Do you have a if script running someware?

A group address with for example 5 lights equals 0 if all lights are out but equals 1 if for example only 1 light is on. If HASS retrieves all states and 1 is on, then the group address with all the lights equals also 1 (1=on).

You could test this by making sure ALL lights are out and the restart HASS... If I'm correct then in this case all lights remain out...

Right?

Julius2342, 2017-10-05:
@TinusK : yes, only group reads.

Do you know if group responses may switch on/off other devices?

It also might be possible that the state of a binary sensor is requested and the answer triggers some lights to be switched on.

TinusK, 2017-10-05:
@Julius2342: With only ETS, no. But if somewhere a script or automation is running, then yes, depending on the script/automation (for example using HASS or some other logic device).

I think one must know how this group address is configured in ets... But I think this is not a problem of xknx or hass.

@mariwing: could you tell me how this group is constructed in ets? (maybe screenshot)

TinusK, 2017-10-05:
Just for info... my entire home is KNX and I also have some group addresses with several lights in it and I do not have this problem.

mariwing, 2017-10-05:
@Julius2342 Home Assistant is running as hass.io so I was under the assumption that I could not install anything extra in there. I have another Rasperry Pi that is setup as a usb-to-knx-ip hub so that's the one Home Assistant is really communicating with. I tried installing and running python on that device.

@TinusK I definitely think that there is something "weird" with my knx setup, this is a setup that was done by a very unexperienced installer ten years ago and during these ten years I have fiddled with ETS and trying to set the system up the way I want it. And now when I try to integrate it with Home Assistant/xknx it is surely some of my mis-configuration that is causing this.

I opened this issue just to get an opinion on what I am seeing is normal or not and it seems that it is not :-) So I will keep on investigating more with the bus monitor, as I've mentioned a lot of messages got fired out on the bus on startup of home assistant and all the read-status-messages. For your information I have around 50 knx devices setup in Home Assistant so I will probably remove all of them and start re-adding them to investigate. Since I have an old but working version of ETS I can reprogram my devices when I find the culprit.

Thanks again guys for all your quick replies and suggestions, closing this particular issue now because I have gotten answer to my question and this seems to be more of a problem with my setup than with xknx.

closed 2017-10-05 by mariwing

Julius2342, 2017-10-06:
If any more problems arise, we are happy to help :)