XKNX/xknx #72: Covers and Wind Alarm

Karssi, 2017-10-05:
Hi Julius,

I just stumbled on a minor cosmetic issue with Covers during a Wind Alarm/Block.
If I issue a down command during a Wind Alarm block HAss will show, both by slider movement and after runtime (calculated reaching of 100% closed) the greying out of the down arrow, that the cover is moving when actually it is locked in the up position.
Would it be possible for the xknx component to look at the "Status of Movement" address (1bit DPT 1.010 start/stop) to determine if the cover is actually moving?

Julius2342, 2017-10-05:
is this status of movement actively pushed?

(to be honest, i actually do not really know how to actively handle this ..)

My covers actively send out the position address when the cover is stopped. This status is used to stop the travel-calculator and move the slider to the current position. Do your covers do the same)

TinusK, 2017-10-05:
The problem here is that hass does not know that the cover is blocked. If you press the down arrow, the knx down command is send to the bus but the cover stays in its current position (movement is blocked) and therefor a position is not returned to the bus (while the cover is blocked, the actor simply ignores the move command).

To solve this, a block address needs to be given to the xknx (e.g. block_address=3/4/5) and if this address equals '1' then all cover buttons needs to be greyed out until the address becomes '0'.

Or maybe an other, more simpler solution is possible but I cant think of it at the moment :)

Julius2342, 2017-10-05:
ah! got it. So the alarm is not issued during movement, it happens some time before and should disable the whole component.

Karssi, 2017-10-06:
Yes, Tinusk is correct. An additional address for Block would be great.
As soon as a cover is blocked (by high wind or low temperature) this will turn to '1'.

marvin-w, 2020-08-26:
@farmio was so kind and opened the necessary architecture issue in home assistant. Right now it's not possible for us to block covers in the frontend. 

We could implement the block address but it wouldn't have any visual impact on HA unless the architecture change gets approved in which case we could proceed here.

Issue: https://github.com/home-assistant/architecture/issues/429

da-anda, 2020-09-28:
would it be possible to already add support for alarm/lock state GAs in XKNX and at least show them as attributes?

PR #654 (2021-04-04): Add support for cover locking (weather alarms etc.)

## Description

This PR adds support for a dedicated `movement_blocked` attribute in HA.

[image]

Fixes #435, fixes #72

## Type of change

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] This change requires a documentation update

## Checklist:

- [x] I have performed a self-review of my own code
- [x] The documentation has been adjusted accordingly
- [x] The [... cut]
closed 2021-04-04 by marvin-w