XKNX/xknx #73: CouldNotParseTelegram should not block whole stateupdater

Julius2342, 2017-10-07:
Current behaviour
-------------------

A CouldNotParseTelegram exception while reading the states interrupts the stateupdater. 

```
File "/config/deps/lib/python3.6/site-packages/xknx/devices/notification.py", line 85, in _process_message
raise CouldNotParseTelegram()
xknx.exceptions.exception.CouldNotParseTelegram: <CouldNotParseTelegram description="" />
```

Expected behaviour
--------------------

* Stateupdater continues after a wrong telegram
* human readable explanation of which payload was wrong and how to fix this.

zimgui, 2017-11-19:
I don't know if this can help but I receive this message when the climate control is activated.

Julius2342, 2018-01-21:
This works now.

closed 2018-01-21 by Julius2342

zimgui, 2018-01-21:
Sorry Julius2342 but I'm still experimenting the same issue:

Log Details (ERROR)
Sun Jan 21 2018 14:51:08 GMT+0100 (CET)

Exception while processing telegram: <CouldNotParseTelegram description="" />
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/xknx/core/telegram_queue.py", line 99, in process_telegram
    yield from self.process_telegram_incoming(telegram)
  File "/srv/homeassistant/lib/python3.5/site-packages/xknx/core/telegram_queue.py", line 130, in process_telegram_incoming
    yield from device.process(telegram)
  File "/srv/homeassistant/lib/python3.5/site-packages/xknx/devices/climate.py", line 286, in process
    yield from self.setpoint_shift.process(telegram)
  File "/srv/homeassistant/lib/python3.5/site-packages/xknx/devices/remote_value.py", line 76, in process
    raise CouldNotParseTelegram()
xknx.exceptions.exception.CouldNotParseTelegram: <CouldNotParseTelegram description="" />

This happen when I change between Operation Modes

Julius2342, 2018-01-21:
Home assistant still includes the old xknx version. PR for HASS to upgrade version and KNX plugin will come soon.