XKNX/xknx #74: KX RGBW / DMX controller

adonno, 2017-10-26:
Ist there any plans on adding this?

I have several LED strips and i would like to chnage their color via home-assitant 

Thnak you

Julius2342, 2017-10-26:
I don't have some ... so it would be somebody else to implement these. Im happy to help.

(my roadmap is: finish climate control, scenes, more DPT types)

adonno, 2017-10-26:
What a shame i hope you aren't the only one working on the KNX implementation, i have no knowledge at all otherwis I would try it

Julius2342, 2018-02-04:
the HASS xknx plugin now supports colors. May you test?

adonno, 2018-02-04:
Hi Julius unfortunately i took fibaro rgbw as there wasn't an imminent solution in sight. i will talk to my electricion to provide me with one and i'll let you know asap 

Sorry tut mir echt leid

moustic999, 2018-08-29:
I successfully tested colors  with the HASS xknx plugin. It works, but it must be able to read the state of the light at the start of Hassio.
in my case, my KNX device didn't had the flag R on the light status, only T(ransmit). adding R has fixed and everything works like a charm.
This issue can be considered as Fixed

closed 2019-04-02 by farmio