XKNX/xknx #77: CouldNotParseTelegram description

adonno, 2017-11-06:
Hi i am having following error message 

any idea what this could come from ? 

2017-11-06 15:29:25 ERROR (MainThread) [xknx.log] Exception while processing telegram: <CouldNotParseTelegram description="" />
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/xknx/core/telegram_queue.py", line 99, in process_telegram
    yield from self.process_telegram_incoming(telegram)
  File "/usr/local/lib/python3.6/site-packages/xknx/core/telegram_queue.py", line 130, in process_telegram_incoming
    yield from device.process(telegram)
  File "/usr/local/lib/python3.6/site-packages/xknx/devices/cover.py", line 229, in process
    position_processed = yield from self.position.process(telegram)
  File "/usr/local/lib/python3.6/site-packages/xknx/devices/remote_value.py", line 76, in process
    raise CouldNotParseTelegram()
xknx.exceptions.exception.CouldNotParseTelegram: <CouldNotParseTelegram description="" />

i also have following error messages but that iss a adress related issue just adding this as perf info in case teh upper error could be linke to the wrong adresses 

2017-11-06 15:24:57 WARNING (MainThread) [xknx.log] Could not read value of <Light name="Lum ext avant" switch="<Address str="1/1/0" />/<Address str="1/1/1" />/None/None" /> 1/1/1
2017-11-06 15:24:59 WARNING (MainThread) [xknx.log] Could not read value of <Light name="Garage" switch="<Address str="1/1/8" />/<Address str="1/1/9" />/None/None" /> 1/1/9
2017-11-06 15:25:00 WARNING (MainThread) [xknx.log] Could not read value of <Light name="WC Cave" switch="<Address str="1/1/10" />/<Address str="1/1/11" />/<DPTBinary value="1" />/Value.ON" /> 1/1/11
2017-11-06 15:25:01 WARNING (MainThread) [xknx.log] Could not read value of <Light name="Lamp ext ARR" switch="<Address str="1/1/14" />/<Address str="1/1/15" />/None/None" /> 1/1/15
2017-11-06 15:25:05 WARNING (MainThread) [xknx.log] Could not read value of <Switch name="Prise Devant" switch="<Address str="1/5/1" />/None/None/None" /> 1/5/1

Julius2342, 2017-11-06:
Could not parse telegram is usually when the wrong type of group address is assigned to a xknx-device. (e.g. xknx expectes a float [2 bytes] but there is a 1 Byte array on the bus.

This problem is related to #73 

The "could not read value of" warn messages usually can be ignored. Sometimes the group address is simply not readable.

Julius2342, 2017-11-06:
regarding problem 1, you should review your configuration. E.g. by commenting out lines to see which group address is broken.

(I want to make the error more readable in the related task)

Problem 2 you can simply ignore.

adonno, 2017-11-07:
ok great thank you. seams to be related to my climate components i am having lot of issues with them

Julius2342, 2018-01-21:
This should also work now. please reopen issue if problem persists.

closed 2018-01-21 by Julius2342