XKNX/xknx #79: Improve requirements.

encbladexp, 2017-11-08:
Currently we use this requirements:
```
pyyaml>=3.11,<4
pylint
netifaces
tox
coveralls
pytest
pytest-cov
pytest-timeout
```

We should improve this a little bit:
* Split requirements.txt in `requirements/base.txt` and `requirements/development.txt`
* Always pin requirements to releases, currently it is easly possible to break things if requirements get updates upstream, also updates on client system get not applied, as the requirement `foobar` is met after installation, only `foobar==X.y.z` would force an update on `pip install -r requirements/base.txt`

I can provide an pull request if required, also i would recommend integration of requires.io, so we have an status on our pinned requirements.

Julius2342, 2017-11-09:
all that makes totally sense ...

encbladexp, 2017-11-09:
I create a Pull Request.

Julius2342, 2017-11-09:
That would be great!

commit e2a51cc0 (2017-11-09): Splitted requirements to dedicated files. Fixes #79

Now we have three different requirements files:
* production.txt, for all users without any development wishes
* testing.txt, with all requirements needed for running tests
* development.txt, with all additional development requirements
closed 2017-11-09 by Julius2342