XKNX/xknx #87: Problem integrating xknx with other asynchronous python code

GeorgFerdinandSchneider, 2018-01-05:
Dear xknx-Team!

thank you for making xknx available! This truly is a great python library for home automation.

We would like to extended the capabilities of xknx and add a RESTful API to it to integrate it with the web framework Twisted Klein.

We use the example provided to build the following code which works fine with our KNX hardware:

```python
"""Example for switching a light on and off."""
import asyncio
from xknx import XKNX
from xknx.devices import Light

async def main():
    """Connect to KNX/IP bus, switch on light, wait 2 seconds and switch of off again."""
    xknx = XKNX()
    light = Light(xknx,
                      name='VarBool',
                      group_address_switch='2/1/1') 
    await xknx.start() 
 
    i = 0
    while i<5:
        i = i + 1
        print ("Turn on, wait 1 s")
        await light.set_on()
        await asyncio.sleep(1)
        print ("Turn of wait 1s")
        await light.set_off()
        await asyncio.sleep(1)
    await xknx.stop()    

loop = asyncio.get_event_loop()
loop.run_until_complete(main())
loop.close()
```

The Klein web framework is a simple asynchronous web framework to host small websites or build your own REST API (Link)[https://klein.readthedocs.io/en/latest/examples/await.html]

In the (documentation)[https://klein.readthedocs.io/en/latest/examples/await.html] the developers of Klein show how to use async/await to add own functions to it. The example code is:

```python

import treq
from klein import Klein
app = Klein()

@app.route('/', branch=True)
async def google(request):
    response = await treq.get(b'https://www.google.com' + request.uri)
    content = await treq.content(response)
    return content

app.run("localhost", 8080)
```

This is our attempt to integrate xknx and Klein:

```python

import treq
from klein import Klein
from xknx import XKNX
from xknx.devices import Light
import asyncio

app = Klein()

@app.route('/on', [ 'GET' ])
async def on(request):
    xknx = XKNX()
    light = Light(xknx,
                  name='VarBool',
                 group_address_switch='2/1/1')
    await xknx.start()
    await light.set_on()
    await xknx.stop()

    return 'Light is on'

@app.route('/off', branch=True)
def off(request):
    xknx = XKNX()
    light = Light(xknx,
                  name='VarBool',
                  group_address_switch='2/1/1')
    await xknx.start()
    await light.set_off()
    await xknx.stop()
    
    return 'Light is off'

app.run("localhost", 8080)
```
Unfortunately this does not work. We tried different things but did not come up with a solution. We usually get an error "error.BuiltIn - No yield from provided for Future".

It would be great if you might suggest how to integrate xknx and other asynchronous python code. We are also willing to contribute this example if this interesting to the xknx documentation.

Best

Georg

Julius2342, 2018-01-05:
uh, no idea. Some thoughts:

* you should create one global XKNX object.  Creating an XKNX object (plus setting up the Tunnel connection each time) is very inefficient - and it wont work if you increase the load.
* You should also instantiate XKNX(loop=klein_loop) with the event loop object used by Klein - using two event loops will potentially fail. (no idea how to access the event loop within klein)

I have to admit i never heard of Twisted / Klein. I'm not sure if it is even compatible[0]. Are there any reasons for using klein instead of e.g. aiohttp?

Best

Julius

[0] https://www.reddit.com/r/Python/comments/3w1w0p/asyncio_vs_twisted_is_it_more_like_27_vs_34/

GeorgFerdinandSchneider, 2018-01-08:
Hi Julius!

thanks for the suggestions. I will check with the Klein developers how to access the loop object. I adapted the code a bit to reflect your suggestions towards single instantiating the xknx object.

I was aware of aiohttp but Klein has the charming interface of Flask. Do you have experience with aiohttp in particular coupling it to xknx?

Best

Georg

Julius2342, 2018-01-08:
XKNX has an integration into [home-assistant](http://home-assistant.io) which uses aiohttp as underlaying web framework. From what I know the interface is also similar  (probably not as similar as Klein) to flask.

koenhendriks, 2018-01-10:
With some help on discord i managed to get the aiohttp server working with xknx.
Here is is an example that can switch lights on and off using a `GET` request to the server.

To turn on the Living room light i can do the following url:
```http://localhost:5000/?action=off&name=livingroom_light```

## config.yaml
``` yaml
groups:
    light:
        livingroom_light: {group_address_switch: '1/0/4'}
```
## server.py
``` python
"""Example for using iohttp to switch a light."""
import asyncio
import datetime
from xknx import XKNX
from aiohttp import web

xknx = None

def log(message):
    log_message = "[" + datetime.datetime.now().strftime("%d-%m-%Y %H:%M:%S") + "] " + message
    log_file = open('log.txt', 'a')
    log_file.write(log_message+"\n")
    return print(log_message)

async def monitor():
    global xknx
    global loop
    xknx = XKNX(config="config.yaml",loop=loop)
    server = web.Server(handler)
    await loop.create_server(server, "0.0.0.0", 5000)
    log("Server running http://localhost:5000")
    await xknx.start(daemon_mode=True)
    await xknx.stop()

async def main():
    log("Starting KNX listener ")
    await monitor()

async def handler(request):
    global xknx
    action = request.rel_url.query['action']
    name = request.rel_url.query['name']
    if(action == 'off'):
        await xknx.devices[name].set_off()
    else:
        await xknx.devices[name].set_on()
    return web.Response(text="OK")

if __name__ == "__main__":
    # pylint: disable=invalid-name
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())
    loop.close()
```

GeorgFerdinandSchneider, 2018-07-25:
Hi!

@koenhendriks your solution based on aiohttp looks neat. Indeed this is an option.

We are able to implement the following code which supposingly works. We still need to test with a real KNX device.

Best

Georg

 ```python
from xknx import XKNX
from xknx.devices import Light
from klein import Klein

loop = asyncio.get_event_loop()
asyncio.set_event_loop(loop)

app = Klein()

@app.route('/on')
async def on(request):
   xknx = XKNX()
   light = Light(xknx,
           name='VarBool',
           group_address_switch='2/1/1')
   xknx.start()
   light.set_on()
   xknx.stop()

   return 'Light is on'

@app.route('/off')
async def off(request):
   xknx = XKNX()
   light = Light(xknx,
           name='VarBool',
           group_address_switch='2/1/1')
   xknx.start()
   light.set_off()
   xknx.stop()
   
   return 'Light is off'

app.run("localhost", 8080)
```

zbikmarc, 2018-12-16:
@GeorgFerdinandSchneider 
Unfortunately is seems to not work.
When code is without `await` in front of `xknx.start()`, `light.set_on()` and `xknx.stop()` there is message like:
```
2018-12-16 17:57:53+0000 [-] klein_webapp.py:17: builtins.RuntimeWarning: coroutine 'XKNX.start' was never awaited
2018-12-16 17:57:53+0000 [-] klein_webapp.py:18: builtins.RuntimeWarning: coroutine 'Light.set_on' was never awaited
2018-12-16 17:57:53+0000 [-] klein_webapp.py:19: builtins.RuntimeWarning: coroutine 'XKNX.stop' was never awaited
```
and telegrams are not sent.

When `await` is added it throws exception:
```
2018-12-16 18:00:47+0000 [-] Unhandled Error
        Traceback (most recent call last):
          File "/home/pi/.local/lib/python3.5/site-packages/klein/app.py", line 227, in _f
            return _call(instance, f, request, *a, **kw)
          File "/home/pi/.local/lib/python3.5/site-packages/klein/app.py", line 52, in _call
            result = ensureDeferred(result)
          File "/home/pi/.local/lib/python3.5/site-packages/twisted/internet/defer.py", line 911, in ensureDeferred
            return _cancellableInlineCallbacks(coro)
          File "/home/pi/.local/lib/python3.5/site-packages/twisted/internet/defer.py", line 1529, in _cancellableInlineCallbacks
            _inlineCallbacks(None, g, status)
        --- <exception caught here> ---
          File "/home/pi/.local/lib/python3.5/site-packages/twisted/internet/defer.py", line 1418, in _inlineCallbacks
            result = g.send(result)
          File "klein_webapp.py", line 17, in on
            await xknx.start()
          File "/home/pi/.local/lib/python3.5/site-packages/xknx/xknx.py", line 68, in start
            await self.knxip_interface.start()
          File "/home/pi/.local/lib/python3.5/site-packages/xknx/io/knxip_interface.py", line 82, in start
            self.connection_config.scan_filter)
          File "/home/pi/.local/lib/python3.5/site-packages/xknx/io/knxip_interface.py", line 98, in start_automatic
            gateways = await gatewayscanner.scan()
          File "/home/pi/.local/lib/python3.5/site-packages/xknx/io/gateway_scanner.py", line 103, in scan
            await self._send_search_requests()
          File "/home/pi/.local/lib/python3.5/site-packages/xknx/io/gateway_scanner.py", line 122, in _send_search_requests
            await self._search_interface(interface, ip_addr)
          File "/home/pi/.local/lib/python3.5/site-packages/xknx/io/gateway_scanner.py", line 138, in _search_interface
            await udp_client.connect()
          File "/home/pi/.local/lib/python3.5/site-packages/xknx/io/udp_client.py", line 162, in connect
            lambda: udp_client_factory, sock=sock)
          File "/usr/lib/python3.5/asyncio/base_events.py", line 949, in create_datagram_endpoint
            yield from waiter
          File "/usr/lib/python3.5/asyncio/futures.py", line 381, in __iter__
            assert self.done(), "yield from wasn't used with future"
        builtins.AssertionError: yield from wasn't used with future

```

Have you solved this problem?

Julius2342, 2018-12-16:
It is better to instanciate a global xknx object instead of creating one for each command. I would suggest you first try to run the code outside a webserver and then add the webserver later on.

closed 2021-01-25 by farmio