XKNX/xknx #94: XKNX UDP Error killing HASS log

FireFrei, 2018-01-16:
Hey, I got a problem :-)

I use xknx with Hass and it is filling up my hass logs with the following error message:

```
...
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "uvloop/cbhandles.pyx", line 52, in uvloop.loop.Handle._run
  File "uvloop/handles/udp.pyx", line 60, in uvloop.loop.UDPTransport._on_read_ready
  File "/usr/local/lib/python3.6/site-packages/xknx/io/udp_client.py", line 59, in error_received
    self.xknx.logger.warning('Error received: %s', exc)
AttributeError: 'UDPClientFactory' object has no attribute 'xknx'
2018-01-16 22:46:20 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback UDPTransport._on_read_ready
Traceback (most recent call last):
  File "uvloop/handles/udp.pyx", line 56, in uvloop.loop.UDPTransport._on_read_ready
OSError: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:
...
```

I just deleted the hass log file with about 65GB of spam. This filled up my servers hard disk :-D

My xknx config in hass is pretty minimalistic:
```
knx:
  tunneling:
    host: '192.168.1.66'
    port: 3671
    local_ip: '192.168.1.60'
```

I've noticed the problem in Hass Version 0.60 and 0.61 

Can you help me with this issue? Cheers :-)

FireFrei, 2018-01-16:
I'm using the official HASS docker image that ships Python 3.6.4 and xknx 0.7.18

Julius2342, 2018-01-17:
Hmm,

what I don't understand, how UDPClientFactory can be initialized without having the `self.xknx` object being initialized:

https://github.com/XKNX/xknx/blob/master/xknx/io/udp_client.py#L42

Any ideas?

(There is also another issue for a better handling of UDP errors ... https://github.com/XKNX/xknx/issues/89 ).

encbladexp, 2018-01-17:
Maybe, this issue has another source, and not the exception we see now. Maybe this exception is the result of an different issue…

janitza-bjag, 2018-02-01:
Hi,

i have the same issue with HA 0.62.1 used in docker with https://hub.docker.com/r/homeassistant/raspberrypi3-homeassistant/ image

The "error" seems to appear randomly after a few hours/days of running HA. When it happens the log file size crashed the docker image ...

any hints appreciated!

Julius2342, 2018-02-04:
Another question: does this problem arise directly after starting HASS? Or does it happen when hass was running for a while?

FireFrei, 2018-02-05:
It happens when hass is running for a while

janitza-bjag, 2018-02-07:
It seems that 0.62.1 is more stable according to this issue, it is running for 5 days now ...

mag2bue, 2018-07-25:
@janitza-bjag you found a solution? I guess I have the some issue. After a few hours/days I lost the KNX connection. HA is still running. 
HA 0.7.4.1

marvin-w, 2018-12-29:
@Julius2342: Please close this issue since no one has experienced this issue anymore since the PR.