XKNX/xknx #101: Error in XKNX binary_sensor after latest update

Karssi, 2018-01-25:
Hi.

Yesterday I updated xknx to the latest version and ever since I have the following error in Home Assistant.

2018-01-25 10:12:42 ERROR (MainThread) [homeassistant.components.binary_sensor] Error while setting up platform xknx
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/helpers/entity_component.py", line 171, in _async_setup_platform
    SLOW_SETUP_MAX_WAIT, loop=self.hass.loop)
  File "/usr/lib/python3.5/asyncio/tasks.py", line 400, in wait_for
    return fut.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/usr/lib/python3.5/asyncio/coroutines.py", line 210, in coro
    res = func(*args, **kw)
  File "/home/homeassistant/.homeassistant/custom_components/binary_sensor/xknx.py", line 68, in async_setup_platform
    async_add_devices_config(hass, config, async_add_devices)
  File "/home/homeassistant/.homeassistant/custom_components/binary_sensor/xknx.py", line 94, in async_add_devices_config
    reset_after=config.get(CONF_RESET_AFTER))
TypeError: __init__() got an unexpected keyword argument 'reset_after'

Everything else works except the binary_sensor.

Any ideas what could cause this? It worked before the update..

Cheers,
Michiel

Julius2342, 2018-01-25:
This is most probably a version mismatch between HASS-plugin and XKNX library.

(reset after was introduced recently)

Karssi, 2018-01-25:
Ok..

I am running the latest HAss v0.61.1
Just wait for an update to HAss or can it be solved before that?

Julius2342, 2018-01-25:
Are you running xknx-custom-plugin or default knx-plugin?

Are you running HASS out of git checkout or installed version?

Karssi, 2018-01-25:
Running the Hassbian 1.31 and updated HASS via "sudo hassbian-config upgrade home-assistant" to v0.61.1. Also using xknx-custom-plugin. Updated by entering the xknx folder and "sudo git pull" and then copying xknx/home-assistant-plugin/custom_components to the homeassistant config folder.

closed 2020-08-18 by farmio