XKNX/xknx #103: There is need for a KNX read-request-service

pholausson, 2018-01-27:
Hi
Some KNX physical sensors don´t send telegrams as often as you would like. Or, sometimes there is need for requesting a status/telegram to be sent by the physical sensor to the bus. 
The current "payload" request only sends knx_write, not read and the sensors dont respond to write-requests.

I had this problem and I found a smart guy that had shared a custom component that works (I have tested and uses this as of now)

https://gist.github.com/ct-23/b707f3c2d8406c8697a25a3ce619f3c2#file-knx_reader-py

It would really improve HA KNX functionality if a similar knx_read-service was implemented. 

Best
/PH

Julius2342, 2018-02-04:
Hi,

wouldn't it be better if you could trigger the read of a sensor directly instead of triggering a read and the value is then received by a different sensor?

```
Domain: knx
Service: sync 
Service Data: {"entity_id": "OutsideTemperature"}
```

pholausson, 2018-02-04:
Hi,
I dont know really.
But the way my KNX-setup is (and most setups, as I understand it), is that
all sensors that you use in HA receive telegrams routed or tunneled that
sensors send to the KNX group addresses they are configured to send their
data to. For example my knx room temperature sensors send to the group
adress assigned to my heat actuator. That way all telegrams are
continuously picked up and presented in HA.

But, some sensors you need to trigger that manually and that is made by a
KNX read-request and the sensor answers to the sensors ETS-configured group
adress in the datatype that it is configured to communicate in.

So I think that the possibility to trigger a common KNX read-request is the
way to go.

Another scenario is if you have sensors that only transmit at long
intervals, if you restart HA or change something, you have to wait for the
sensor to send its telegram (sometimes hours....). If you could trigger a
KNX read, you could force-update all sensor data at restart or other demand.

Best
/PH

*Per-Håkan Olausson*
Leg läkare, Dermatolog, MBA, Verksamhetschef

Tel: 08-55 00 35 70 | Fax: 08-55 00 35 79
e-post: kontakt@skindoc.se

*Postadress:* SkinDoc, Svärdvägen 11, 18233 Danderyd
*SkinDoc* hudläkarmottagningar: www.skindoc.se

farmio, 2019-04-09:
@derandiunddasbo I cannot reproduce this behavior. When I restart my HA (0.90.2 at the moment but it has been this way since I use HA - ~1 year) every configured device is read from the bus. sensors, binary_sensors and other devices alike.

The sensor `address` this is indeed misleading wording. A Sensor class is a read only device in xknx and therefore has only one address. This is [internally mapped](https://github.com/XKNX/xknx/blob/40888c909e890f4be70d5ec7d94949ae30a2298a/xknx/devices/sensor.py#L39) to the `group_address_state` attribute of the underlying RemoteValue class. 
So the `address` of a sensor is indeed a `state_address`

derandiunddasbo, 2019-04-23:
Sorry for the late reply, but I was on vacation the last weeks and couldn't dig into this anymore.

There have been some issues with set and state addresses for climate components in the past, thus I had to read operation modes and temperature values with my knxtool workaround as well. But since the address / state_address fix they are updating fine on startup. 

Today I monitored all values read from the bus by HA during restarts in ETS5 and matched them to all configured knx entities. Turns out, that _all_ state addresses from _all_ KNX entities are read on startup, _except_ for _all_ configured KNX binary sensors.

KNX binary sensors never worked. Even long before I started to modify the xknx component. While I've been updating my HA container today I tested this without and with my modifications for DPT 16.000 sensors. These mods don't affect the issue in any way. The issue with HA using component files from my config dir is resolved, since I renamed the folder.

The config for these sensors is dead simple:

```
- platform: knx
  name: DG Bad Kontakt Fenster oben
  address: '1/3/11'
```
I tried to omit optional values, i.e. the name, or add values that should be optional, f.e. the device class, but that doesn't change the behavior.

When issuing read requests in ETS or with knxtool, HA updates the sensors immediately when it sees the GroupResponse packets on the bus. But it never issues read requests for binary sensors by itself, neither on startup nor sheduled.

closed 2019-05-01 by farmio

derandiunddasbo, 2019-06-13:
Just curious: How and when do changes and fixes for xknx get updated in the home assistant plugin? Besides this fix for state restoration of knx binary sensors, I'm also hoping to see some enhancements getting incorporated into the HA plugin, especially this: https://github.com/XKNX/xknx/issues/116

But in the HA change logs for the last couple of versions xknx never appears. :-)

Lilith5th, 2020-06-11:
was there any resolution for the matter?
I still get sensors that dont post as often enough, or dont give a reply on "read" command. 

I've tried using knxtool read, groupread, and vbusmonitor1 to get the data... but nothing gives me the result.

farmio, 2020-06-11:
If they don't respond to read requests we can't do anything about it - this has to be configured in ETS. 
Binary sensors are read by now on init an the regular 1h Intervall that all sensors in xknx are Read currently. 
You can also manually trigger a read request through the  homeassistant.update_entity service.

PR #199 (2019-05-01): read binary sensors on startup
- rebased HA plugin from HA 0.92.0 (mainfest.json)
- removed group_address form BinarySensor in favor for group_address_state
- replaced group_address in Sensor with group_address_state

Sensor and BinarySensor now have `group_address_state` as their only group address attributes as they are read only devices. With this change binary sensors are read on startup ( #140 ).

This resolves #103