XKNX/xknx #105: position_state_address fails with old KNX devices

MrChrisCool, 2018-01-28:
Hi !

My Knx cover device is [Hager TS225](https://www.hager.it/files/download/0/954_1/0/TS225_TS226.pdf) with application program [TK338](https://www.hager.pt/files/download/0/2901_1/0/TS223_225_ETS_FR.pdf)

I have the following error in home-assistant.log

_2018-01-28 18:06:47 ERROR (MainThread) [xknx.log] Exception while processing telegram: <CouldNotParseTelegram description="" />
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/xknx/core/telegram_queue.py", line 99, in process_telegram
    yield from self.process_telegram_incoming(telegram)
  File "/srv/homeassistant/lib/python3.5/site-packages/xknx/core/telegram_queue.py", line 130, in process_telegram_incoming
    yield from device.process(telegram)
  File "/srv/homeassistant/lib/python3.5/site-packages/xknx/devices/cover.py", line 229, in process
    position_processed = yield from self.position.process(telegram)
  File "/srv/homeassistant/lib/python3.5/site-packages/xknx/devices/remote_value.py", line 76, in process
    raise CouldNotParseTelegram()
xknx.exceptions.exception.CouldNotParseTelegram: <CouldNotParseTelegram description="" />_

It's because position_state_address in xknx is of type 5.001 and my old knx device send its position state with type 1.001 (open or close...)

The question is :
Is it possible to modify position_state_address to handle type 1.001 and 5.001 ?
for example 00(1.001) -> 0%(5.001) and 01(1.001) -> 100%(5.001)

Many thanks !
Chris

MrChrisCool, 2018-01-28:
Hmm... I just see that Issue #68 should resolve this one...

jncanches, 2018-02-07:
Is there any news forthis issue ?
I am in the same case with a cover state type in 1.001

Many thanks !

JN

Julius2342, 2018-02-07:
Not really, Im not happy with #107 . We should solve this the right way and introduce the correct DPT abstraction to handle these devices.

Julius2342, 2018-02-07:
@jncanches : may you jump in and implement it? :-)

MrChrisCool, 2018-02-08:
Hi jncanches, No news for this issues, but you can use the hack #107 to solve your problem, it will not break your covers as a 1.00x can always be converted to a 5.00x value.

But Julius2342 is right, we must provide a correct implement.

Further more, this old devices that sending position in 1.00x type introduce the state unknown the type 5.00x should not have.
for exemple 01(1.00x) is closed and 00(1.00x) is open/unknown that mean only one position is sure...
that mean changing the code that calculate the position.

Something great would be in the config file the possibility to do :

- platform: xknx
  name: MainOffice Cover
  move_long_address: 1/3/81, 1.001
  move_short_address: 1/3/85, 
  position_state_address: 1/3/91, 5.001

property: addressValue, typeValue

if typeValue is not present use a default value.

The long way...

farmio, 2020-08-18:
There is no binary_position_state_address, but telegrams to move_long_address are now (0.12.0) processed by xknx, so it should be in sync (if the actuators and xknx movement times match).

PR #322 (2020-08-13): Cover update - up/down from bus; stop and movement direction
this is derived from #259 so 
- in xknx 0% is open and 100% is closed now
- adds is_opening and is_closing
- react to up/down from bus and start travel calculator
- add group_address_stop for actuators that have it (or don't support step)

resolves #314 
fixes #257 (I hope)
fixes #105 (without implementing the DPT1 state address - just listening for the normal group_address so initial status may be off)
closes #25
closed 2020-08-18 by farmio