XKNX/xknx #113: Add DPT 13.013 Active Energy (kWh) - 4 Byte Signed Value

TinusK, 2018-02-20:
I have a Power Measuring KNX Device hooked up to my electrical grid. It measures mainly:
- Current
- Voltage
- Power
- Active Energy

As far as I know, the 'Active Energy' DPT is not implemented yet. It is the value that is used to measure what Electrical Energy is used and is used to calculate what you have to pay the electric company :)

I would have added it myself, however my Python knowledge and skills are none existent :(

Thanks!
Tinus

tillsc, 2018-09-28:
Have you tried using `DPT-13` as value_type (or `type` in HA)?

TinusK, 2018-09-29:
No, not yet. Don't see it as a type in the knx sensor component document on the HA website. Has it been added?

I haven't seen any knx update in the update logs of HA. I'm currently running the latest HA 0.78.3... which xknx version is in there?

tillsc, 2018-09-29:
I created a pull request yesterday: https://github.com/home-assistant/home-assistant.io/pull/6374 

As far as I see this will work with older HA / xknx versions too

tillsc, 2018-09-29:
You can find a complete list of supported types here: https://github.com/XKNX/xknx/blob/master/xknx/devices/remote_value_sensor.py#L22-L55

TinusK, 2018-09-29:
Thanx!

closed 2019-04-02 by farmio