XKNX/xknx #116: DPT 16.000 Character String (ASCII)

TinusK, 2018-03-03:
It would be nice to be able to have a sensor type for strings. I would like to be able to display Diagnosis text messages from, for example, my heating actuators.

Ragards,
Tinus

derandiunddasbo, 2019-03-14:
I'd like to back this feature request. My alarm system outputs DPT 16.000 messages which I'd like to process in HA.

Is there any chance, this DPT gets implemented?

farmio, 2019-03-15:
The base is already here: [DPTString](https://github.com/XKNX/xknx/blob/master/xknx/knx/dpt_string.py).
If you know (or want to learn) some python you can try to extend [RemoteValueSensor()](https://github.com/XKNX/xknx/blob/master/xknx/devices/remote_value_sensor.py) and  the HA component.

derandiunddasbo, 2019-03-18:
Well, I "hacked" this into my local installation and it's working fine for me. But even it's only a couple of additional lines of code I have no idea if my implementation meets any requirements (naming, style, etc.) of the xknx project. And TBH I never opened a PR on Github so far. :-)

So I could contribute this, but a little help how to do this would be much appreciated.

farmio, 2019-03-18:
You can just fork the project, commit your changes and file a PR. We can have a look at the diff then. 
For a discussion either is fine. You can also join the discord channel (you'll find a link on xknx.io i think)

derandiunddasbo, 2019-03-22:
Sorry, didn't have time this week to work on this. But IIRC, DTP 16.000 will be provided with fix #189? Thus I don't need to contribute my own fix anymore?

farmio, 2019-03-22:
Right.

closed 2019-04-18 by marvin-w