XKNX/xknx #118: Feature Request: Monitor Scene messages broadcasted on the bus

mar-schmidt, 2018-03-09:
It is desirable to monitor for Scene messages that are broadcasted on the KNX bus. In my example I have wall switches in the bedroom and hallway which triggers KNX scenes (turn off all lights in the house). If I had the possibility to monitor and act on that message I would be able to automate new series of actions in Home Assistant that are triggered by this Scene message. In my example I would like to turn off my Sonos speakers, close the window covers and turn off all my Hue lights.

This was also discussed in Home Assistant forums: https://community.home-assistant.io/t/listening-to-knx-scene/45885/7

iditude, 2018-03-24:
I support this. I have hue lights as well and would love my KNX switch to not only turn everything off, but also inform Home assistant to turn off hue lights (through an automation for instance).

Thanks

farmio, 2018-10-24:
You could configure a sensor in your home assistant setup with the group address of the scene trigger and type `pulse`. This sensor will yield `scene-number - 1` you can act on in your hass.

Julius2342, 2018-10-24:
The most straightforward approach would be to completely replace KNX scenes by HASS scenes. I have done this at my home and it works perfectly :-)

farmio, 2018-10-24:
This may not be easily possible if one has a running KNX installation but no ETS Licence or other visualization / logic servers like a Gira X1 / Homeserver or an Edomi server where scenes are used.

iditude, 2018-11-04:
> The most straightforward approach would be to completely replace KNX scenes by HASS scenes. I have done this at my home and it works perfectly :-)

I second that (for those that have KNX license and can change their setup), but now we need a sensor that monitors a knx button switch. Do you have examples on how to do that?

farmio, 2018-11-04:
Just use a 'binary_sensor' with platform 'knx' and a GA for each button. 
 https://www.home-assistant.io/components/binary_sensor.knx/
I think this approach is against the knx philosophy to have a decentralized/serverless system. But that's my opinion.

iditude, 2018-12-05:
Hello,

I'm sorry but still running into issues with this setup. I'm just trying to capture if a message is send to the Scene GA.

Here's my setup
```  - platform: knx
    name: SceneOff
    friendly_name: "KNX Scene Off"
    address: '3/0/0'
    automation:
      - hook: 'on'
        action:
          - service: light.turn_off
            data:
              entity_id:
                - light.xxx
      - hook: 'off'
        action:
          - service: light.turn_off
            data:
              entity_id:
                - light.xxx
```

This is throwing the following error every time a scene button is pressed:
```
2018-12-05 14:16:53 ERROR (MainThread) [xknx.log] Error while processing telegram <CouldNotParseTelegram description="invalid payload" device_name="SceneOff" payload="<DPTArray value="[0x0]" />"/>
```

Also I know this is suboptimal as this triggers regardless of what scene number is triggered. Ideally I'd like to be able to read the value and only trigger once the right scene is called. 

Any help?

Thanks

farmio, 2018-12-05:
@iditude 
- what kind of HA entity is this? sensor, switch, binary_sensor, light...
- what DPT is your '3/0/0' ? Do you have an ETS installation so you can monitor whats actually sent on the bus on that address?

PS. i'm not sure if this xknx feature request issue is the right place for support questions on home assistant.

iditude, 2018-12-05:
Hello,

I'm trying to implement what has been discussed here. This is a binary switch.

As far as DPT, I'm using the standard scene DPT scene control 18.001 for the receiving end and DPT scene number 17.001 for the devices reading.

Thanks

Julius2342, 2018-12-05:
@iditude : how do you plan to implement this?

I could think of the following approaches:

* make it a general knx-component feature, compared to fire_event_filter. You would register scene knx-addresses and then you can watch these events from automations etc.
* Add a flag to binary_sensor component. (within xknx i would prefer that you improve the scene device and instanciate a scene device.
* Hack something within scene/knx.py, but im not sure if that matches with the concept of hass-scenes.

Tell me where i can help.

Julius2342, 2018-12-05:
(btw did you try `fire_event_filter` to solve your probelm?

iditude, 2018-12-05:
Hello,

Sorry I should have been clearer. I am not trying to implement this (as in coding) as this goes way beyond my capabilities. I was just trying to use whatever is at my disposal to have a way of having HASS react to a scene triggered within the KNX bus and components. 

As far as fire_event_filter is concerned, I'm not yet there as I still don't know how to read the event that is of interest to me (with invalid payload error).

I think the binary sensor maybe the wrong route. If I were to be able to setup a sensor with the GA of the scene, I could react to it through automation. The problem is that I tried and I have no sensor created. 

Thanks

Julius2342, 2018-12-05:
so, using fire_event_filter, you get an invalid payload error? if so it would be easy to fix.

farmio, 2018-12-05:
@iditude it seems you are trying to use a binary_sensor entity (1 bit) on a knx scene address which is probably DPT 18.001 (8 bit - 1 controll, 1 reserved, 6 for the scene number).
try switching the binary_sensor to a sensor of type 'pulse'. Im not sure about 'hook' as I couldn't find it in HomeAssistant documentation. Maybe this will work:
```
sensor:
  - platform: knx
    name: SceneOff
    friendly_name: "KNX Scene Off"
    address: '3/0/0'
    type: 'pulse'

automation:
  - trigger
    platform: numeric_state
    entity_id: sensor.sceneoff
    action:
      - service: light.turn_off
        entity_id:
          - light.xxx
```
After pushing your button you should notice the sensor value - this is the scene number (or scene number -1 if you will) that is emitted on the bus. 
This could cause problems when restarting HA because xknx will want to read the state - if your knx isnt provisioned correctly your lights may turn off after a restart because some device answered with a scene number. Its worth a try.

Julius2342, 2018-12-05:
https://www.home-assistant.io/docs/automation/trigger/ <- this might also be worth reading.

iditude, 2018-12-05:
Hello,

Thanks a lot for your support. I was missing the pulse piece to have HA recognised the sensor. This is much more elegant and works well. 

Last question in the automation (and sorry if this isn't a xKnx direct issue). Values in GA are not updated after a Scene is called (i.e. if we call scene number 1, the GA keeps the value 1).

Hence if one wants to retrigger the same scene, Hass sees no update and the automation doesn't work. 

Any brilliant idea to solve this (i.e. it works only once)?

thanks!

Julius2342, 2018-12-05:
you should try fire event - this does not have any state.

iditude, 2018-12-08:
Hello,

Thanks to all for your support. For posterity, or anyone looking at doing this, this is the full picture (took me a bit of time to understand how events work).

1. Setup Events for KNX (add events fire as well as indicate which address you will monitor)

```knx:
  tunneling:
    host: '192.168.1.108'
    port: 3671
    local_ip: '192.168.1.5'
  fire_event: True
  fire_event_filter: ["3/0/0"]
```

2. setup sensor for the scene GA used in the KNX system

```sensor: 
 -  platform: knx
    name: SceneKNX
    address: '3/0/0'
    type: 'pulse'
```

3. Setup Automation to trigger when an event is raised and then check if the scene number is the right one (through an additional condition, in my example here I only react to scene 0)

```
- alias: SceneAllOff
  hide_entity: True
  trigger:
    platform: event
    event_type: knx_event
    event_data:
      address: "3/0/0"
  condition:
    - condition: state
      entity_id: sensor.sceneknx
      state: '0'
  action:
    - service: light.turn_off
      entity_id:
          - light.xxx
```

Cheers

Julius2342, 2018-12-08:
I wonder if that would be possible without that additional sensor. (dont know how, but i assume that there is a way off accessing the event data/payload somehow, otherwise it would not make sense to have these payloads).

If, not, i think we can close this issue?

farmio, 2018-12-08:
Could we make a sensor of type 'scene' with the fire_event property set by default? This would be clearer to set up. 
And we could add 1 by default to match KNX scene numbers.

iditude, 2018-12-09:
Hello,

I second a more elegant solution than the fire event way but also agree on what has been asked in this issue is resolved.

If someone can code a sensor of type scene that would be awesome and simplify the settings quite a bit. I don't have the knowledge to do so though.

Thanks to all

farmio, 2019-05-29:
A `scene_number` sensor is now implemented with #189 and refined with #198

closed 2019-05-29 by farmio