XKNX/xknx #119: Could not Parse Telegram from setpoint_shift_state_address

zimgui, 2018-03-12:
Hello, I get this error when "setpoint_shift_state_address" is set:

Log Details (ERROR)
Mon Mar 12 2018 19:20:03 GMT+0100 (CET)

Error while processing telegram <CouldNotParseTelegram description="payload invalid" device_name="clCalefaccio" group_address="9/2/2" payload="<DPTArray value="[0x0,0x0]" />"/>

Is using this element a 2 bytes temperature telegram?

In the next image you can see the KNX config of my climate actuator:

[image]

And here the climate configured in HASS:

  temperature_address: 9/2/20
  target_temperature_address: 9/2/0
  setpoint_shift_address: 2/0/3
  setpoint_shift_state_address: 9/2/2
  operation_mode_address: 2/0/9
  operation_mode_state_address: 9/2/3

I don't know if I'm doing something wrong.

Thanks

zimgui, 2018-03-12:
I tried another configuration without setpoint_shift_state_adress and this is the result:

2018-03-12 21:42:13 WARNING (MainThread) [xknx.log] Could not read value of <Climate name="clCalefaccio" temperature="GroupAddress("9/2/20")/None/<DPTArray value="[0x7,0x44]" />/18.6"  target_temperature="GroupAddress("9/2/0")/None/<DPTArray value="[0x7,0xd0]" />/20.0"  setpoint_shift="GroupAddress("2/0/3")/None/None/None" setpoint_shift_step="0.5" setpoint_shift_max="6" setpoint_shift_min="-6" group_address_operation_mode="GroupAddress("2/0/9")" group_address_operation_mode_state="GroupAddress("9/2/3")" group_address_controller_status="None" group_address_controller_status_state="None" /> 2/0/3

I don't known if this can help

farmio, 2019-05-29:
Hi! I don't know if this problem still persists as the issue is nearly 1 year old. 
Have you tried the config of your first post (with setpoint_shift_state_address set) with a recent version of home assistant? 
From the error it looks like your actuator sent you an empty telegram. This should not happen.
For the second post here: This behaviour will be deprecated soon as only *_state_address are read from the bus.

closed 2020-08-18 by farmio