XKNX/xknx #120: Could not read KNX Value

zimgui, 2018-03-12:
Hello again,

Every time I start HASS a continuos Warning about KNX lights is displayed in my log:

Log Details (WARNING)
Mon Mar 12 2018 19:10:37 GMT+0100 (CET)

Could not read value of <Light name="lgHSimple" switch="GroupAddress("0/5/0")/None/<DPTBinary value="0" />/Value.OFF" /> 0/5/0

Is very curious because in every start the KNX light that can't be read is different, one time is the showed in the warning below and another times is any other on my 20 lights.

Any help would be welcomed.

Thanks a lot, you are doing a great job, congratulations!

marvin-w, 2018-12-29:
Is this issue still relevant? I personally haven't experienced this.

greberg, 2018-12-29:
Yes it is still relevant. I get it all the time.

Julius2342, 2018-12-31:
hmm, may you add a delay of e.g. half a second to `xknx/devices/devices.py`: 

```python
    async def sync(self):
        """Read state of devices from KNX bus."""
        for device in self.__devices:
            await device.sync()
            await asyncio.sleep(0.5)
```

(it looks like your knx network gets overloaded and then drops packets).

greberg, 2018-12-31:
Will check when I get back home.

greberg, 2019-01-07:
Hi Julius2342

Tried it out and it seems to work. Haven't seen any warning in the logs since I added 0.5 sleep
Will this be pushed into a release?

Best regards
Peter

Julius2342, 2019-01-07:
@greberg : ok this is good news :)

I remember i had the same problem at home. My Router (Gira) also swallowed packets if I sent too many. That's why i added a rate limit: https://github.com/XKNX/xknx/blob/master/xknx/core/telegram_queue.py#L72 . The limit of 20 per second was more or less a random pick.

May you try to revert your change and try to slightly increase the value? Maybe we find a value, with which also @farmio could live with :) or we make it configurable.

greberg, 2019-01-07:
Im using probably the same amount of devices over tunneling. Hager and Zennio devices and I doubt the topology is wrong. Dummy devices to all adresses? Not really sure what you mean. I can log the bus, can't see and specific error on the bus.

I will try 0.1 to start with, reporting back tomorrow

Best regards
Peter

zimgui, 2019-01-07:
Great news! Make it configurable will be really welcomed, thanks everybody for your help and for improving this amazing software.

marvin-w, 2019-01-07:
@Julius2342 I think making it configurable would be best. I can take a look next week if no one else has done it before :)

greberg, 2019-01-09:
0.2 work out great. Of course configurable would be the preferred solution.

greberg, 2019-01-27:
Did we take a decision to make this customizable? Just for my interest, otherwise I will put some work to script it every time. A lot of releases from HA this days which mean a lot of editing in the docker file :-)

marvin-w, 2019-01-27:
We will make it configurable

greberg, 2019-01-27:
Super awesome! You rock

resoai, 2019-03-09:
Would it be possible to include`rate_limit` also not only in config file but as a configuration variable? I also agree that it would probably make sense to represent it as number of request per second.

farmio, 2019-03-09:
@resoai I think you are talking about home-assistant configuration. This would already be merged in the dev branch so it will hit HA with upcoming 0.90.0 release in about 2 weeks. (see https://github.com/home-assistant/home-assistant/pull/21541 )
It was implemented as requests per second. You will be able to use it like
```yaml
knx:
  rate_limit: 15
```

greberg, 2019-04-01:
Hi.

Still after rate_limit was implemented I get the same errors. Running rate_limit as low as 1 and still gets warning "Could not read value of". Adding the await asyncio.sleep(0.2) into the xknx/devices/devices.py as previously was explained work and takes away the warnings. Any thoughts?

Best regards

ps-tfriedl, 2019-04-13:
Same problem here. I already tried to limit the rate_limit but after restarting hass.io the knx states are incorrect. Didn't try the delay as I'm not sure how to do this with Hass.io

farmio, 2019-04-14:
`rate_limit` prevents the `xknx.telegrams` TelegramQueue() to be drained too quickly to not overload the knx bus. (Only outgoing telegrams apply a pause)

```
async def sync(self):
        """Read state of devices from KNX bus."""
        for device in self.__devices:
            await device.sync()
            await asyncio.sleep(0.5)
```
This prevents all devices to start its `sync()` method at the same time. When a device `sync()` a ValueReader Object is instantiated and its `read()` method is called:
```
async def read(self):
        """Send group read and wait for response."""
        cb_obj = self.xknx.telegram_queue.register_telegram_received_cb(
            self.telegram_received)
        await self.send_group_read()    # adds the GroupValueRead Telegram to the xknx.telegrams TelegramQueue()
        await self.start_timeout()    # defaults to 1 second
        await self.response_received_or_timeout.wait()
```

Now if the internal timeout of the `ValueReader` (1 second) is over before the GroupValueResponse telegram is received from the bus you will get your error `Could not read value of ...` regardless if the GroupValueRead telegram was actually sent to the bus or is still in the `xknx.telegrams` Queue waiting for its turn - wich can be longer than 1 second if there are a lot of state_addresses to read.

@greberg @ps-tfriedl to prevent this issue now you should not set the `rate_limit` lower, but higher to process the `xknx.telegrams` Queue more quickly. Could you try a limit of 30 or 40 and report back if it works better? I think 60 should be about the maximum a KNX-TP can handle.
A second way would be to increase the default `timeout_in_seconds` of 1 in xknx/core/value_reader.py:21

@Julius2342 maybe we should try to apply back-pressure from xknx.telegrams with `maxsize` so `put()` will block until a free slot is available - but I'm not sure how this would affect incoming telegrams. A separate outgoing and incoming queue comes to my mind ;)

ps-tfriedl, 2019-04-18:
How would I observe the queue?

farmio, 2019-04-18:
A quick and dirty way would be to place something like
```
self.xknx.logger.warning("Current Queue size: %s", self.xknx.telegrams.qsize())
```
before https://github.com/XKNX/xknx/blob/b26880ed38190ff4aa844fb4f9a66f5496dc4abd/xknx/core/value_reader.py#L51
This would show you the queue size before putting the next element in the queue. Incoming and outgoing telegrams are in the same queue, but only outgoing trigger the rate_limit sleep. 
If your qsize is 5 or above when putting the next item it is likely that the timeout outraces the rate_limit sleep.

farmio, 2021-01-10:
I think this should be solved by now.

PR #178 (2019-02-18): Make rate limit configurable
Make rate limit configurable.

Leave default to 20 packets per second.

Closes #120
commit d985ea41 (2019-02-18): Make rate limit configurable

Closes #120
closed 2021-01-10 by farmio