XKNX/xknx #126: issues with climate control

adonno, 2018-06-25:
I am strugeling with my climate control 

this is what has been set by my integrator :
  
`#Etage.Heizung.3/4/0	Temperaturmesswert Eltern	Uncertain (2 Byte)	Low	
#Etage.Heizung.3/4/1	Sollwert Komfort Eltern	Uncertain (2 Byte)	Low	
#Etage.Heizung.3/4/3	Betriebsartvorwahl Eltern	Uncertain (1 Byte)	Low	
#Etage.Heizung.3/4/4	DPT_ HVAC Status Eltern	Uncertain (1 Byte)	Low	
#Etage.Heizung.3/4/5	DPT_ RHCC Status Eltern	Uncertain (2 Byte)	Low	
#Etage.Heizung.3/4/6	Betriebsart Komfort Eltern	EIS 1 'Switching' (1 Bit)	Low	
#Etage.Heizung.3/4/7	Betriebsart Nacht Eltern	EIS 1 'Switching' (1 Bit)	Low	
#Etage.Heizung.3/4/2	Aktueller Sollwert Eltern	Uncertain (2 Byte)	Low	
#Etage.Heizung.3/4/8	Betriebsart Frostschutz Eltern	EIS 1 'Switching' (1 Bit)	Low	
#Etage.Heizung.3/4/9	Frostalarm Eltern	EIS 1 'Switching' (1 Bit)	Low	
#Etage.Heizung.3/4/10	Hitzealarm Eltern	EIS 1 'Switching' (1 Bit)	Low	
#Etage.Heizung.3/4/11	Sollwertverschiebung Eltern	EIS 1 'Switching' (1 Bit)	Low	`

here is what i have in my config : 

`  - platform: knx
    name: eltern
    temperature_address: '3/4/1' #3/4/0
    target_temperature_address: '3/4/2'
    setpoint_address: '3/4/11'
    operation_mode_address: '3/4/3'
#    controller_status_address:
    controller_status_state_address: '3/4/4'
    operation_mode_frost_protection_address: '3/4/8'
    operation_mode_night_address: '3/4/7'
    operation_mode_comfort_address: '3/4/6'`

hand here is what i get in my log 

`2018-06-25 10:20:38 WARNING (MainThread) [xknx.log] Could not read value of <Climate name="eltern" temperature="GroupAddress("3/4/1")/None/<DPTArray value="[0x0,0x0]" />/0.0"  target_temperature="GroupAddress("3/4/2")/None/<DPTArray value="[0xc,0x1a]" />/21.0"  setpoint_shift="None/None/None/None" setpoint_shift_step="0.5" setpoint_shift_max="6" setpoint_shift_min="-6" group_address_operation_mode="GroupAddress("3/4/3")" group_address_operation_mode_state="None" group_address_controller_status="None" group_address_controller_status_state="GroupAddress("3/4/4")" /> 3/4/3
`

What did i miss ?

mag2bue, 2018-07-25:
#Etage.Heizung.3/4/11 Sollwertverschiebung Eltern EIS 1 Switching (1 Bit) Low -> should be 1 Byte! 
But not each actuators can 1 byte. I have the issue, that my actuators can only 1 bit or 2 bytes.

climate:
   - platform: knx
     name: HASS-Kitchen.Temperature
     temperature_address: '5/1/1'
     setpoint_shift_address: '5/1/2'   - >  you have  -> setpoint_address: '3/4/11'
     setpoint_shift_state_address: '5/1/3'
     target_temperature_address: '5/1/4'
     operation_mode_address: '5/1/5'

for more details see: https://www.home-assistant.io/components/climate.knx/

adonno, 2018-08-09:
@mag2bue 

Sorry but i'm quite inexperienced with that stuff, how can i solf the issue  "for Dummies" the installation has been done by my electrictian does he need to change anything in that regard ?

moustic999, 2018-08-29:
I have similar issue at home. I use KNX device : Theben HMG 6 T. The manual setpoint offset object is 2 bytes, type : DPT 9.002. 
Acutlly it dos not work properly with xknx, I get error :  "<CouldNotParseTelegram description="payload invalid" .... >

Julius2342, 2018-08-29:
Im more and more convinced, that we need different climate devices for different models. They are so different...

moustic999, 2018-08-29:
look at this doc : http://www.zennio.fr/download/manual_z41_lite_pro_com_en
on page 98, explanation regarding the different shift mode ( absolute / relative 1bit /relative 2 bytes )

basically there is 3 possible modes, but which one are implemented currently in xknx is not clear.
Does someone has a working setup ?

moustic999, 2018-08-30:
I tested again and you are right.
there is 2 issues : 
  - increase button in home assistant does not work !
  - xknx interprets the setpoint_shift data as 1 byte where it is 2-bytes.

I will post logs later on this day.

marvin-w, 2018-12-30:
@moustic999 : at least the increase button should work in the next HA release of you set max and min_temp according to the docs when you're device doesn't support setpoint shift calculations.

G8YTZ, 2019-08-09:
0.97.0 has an issue or breaking change. Basically only 3 operation modes are displayed and they don't work. I think I might have to add an attribute "preset_mode" or "hvac_mode" bit in the documentation it is not clear how this is done or an example. Anyone able to help? Was fine in 0.95.4
[image]

Dudu67, 2019-09-04:
I do have the same problem.
and the on/off switch doesn't show anymore when you open the detail windows ...

kosztag, 2020-02-25:
Hi,
I have problem with Siemens Room Control Unit UP 227/11.
Type of setpoint_shift_address is 9.002, what is not conform with 6.010.

farmio, 2020-02-25:
@kosztag You can try the xknx custom component (see xknx.io ) and clone this PR instead of master: https://github.com/XKNX/xknx/pull/251 
It is a Draft but I think it should work. Maybe you can leave feedback there if it works or if you encounter issues.

kosztag, 2020-02-25:
@farmio type is working. I recognized another issue.
When I change target temperature in UI, shift value is not correct. 
Setpoint temperature: 22 
I would set 23, and the setpoint_shift value is not correct, but sometimes yes.
Requested temperature is not updating.

kosztag, 2020-02-26:
@farmio  It turned out it was my fault. I've set base setpoint for target temperature, not the nominal one.
Now it works perfectly with Siemens Room Control Unit UP 227/11.

One ramark: when I set 0.4 for shift, in bus diagnostic 0.39K shown.

closed 2020-08-18 by farmio