XKNX/xknx #129: Exposing boolean values to the KNX bus

magenbrot, 2018-08-09:
Hi,

I couldn't find a way to expose boolean values to the KNX bus (DPT 1.002). I'm using home-assistant. I would like to have that feature for my door and window contacts (opened=1/closed=0) or to tell my devices if it's day (=1) or night (=0).
Maybe it's already possible, but I couldn't figure it out?

Best regards
Oli

magenbrot, 2018-08-09:
I just noticed that there are also a few DPTs for sensors missing, for example 9.008 part/million (ppm). How can those be added?

spali, 2018-10-18:
Same situation.
Currently I read about a workaround I'll still have to try (create a knx switch for this in HA) and update it with an automation).
But would be great if expose type could be binary to expose a binary_sensor without hacking around with automation etc.

amp-man, 2018-11-02:
This would open up so many possibilities and should not be too hard to implement ;)

@spali @magenbrot  You can also just create an automation and send the values directly to knx.
For the payload you might have to use a jinja data template to convert hass state to boolean.
```
automation:
- alias: Expose Boolean
  trigger: 
     platform: state
     entity_id: sensor.mute
  action:
       - service: knx.send
         data:
           address: "1/1/15"
           payload: 0
```

closed 2019-04-18 by marvin-w