XKNX/xknx #130: Set KNX/IP gateway address

MaxXor, 2018-08-15:
Hello, how can I set an IP address for a KNX/IP gateway? It's not found automatically. Also does the telegram monitor support non-group telegrams?

Julius2342, 2018-08-23:
just define a connection config:

https://github.com/XKNX/xknx/blob/master/xknx/xknx.py#L63

marcor4, 2019-02-15:
Hi! How the connection config can be defined? Could you be a little bit more specific?

farmio, 2019-02-16:
It could look like this:
```python
    xknx = XKNX()
    connection_config = ConnectionConfig(
        connection_type=ConnectionType.TUNNELING,
        gateway_ip="192.168.1.2", gateway_port=3671,
        local_ip="192.168.1.109")
    await xknx.start(connection_config=connection_config)
```

marcor4, 2019-02-19:
I am currently trying the code listed in the previous message and it works:
`  xknx = XKNX()
    connection_config = ConnectionConfig(
        connection_type=ConnectionType.TUNNELING,
        gateway_ip="192.168.1.2", gateway_port=3671,
        local_ip="192.168.1.109")
    await xknx.start(connection_config=connection_config)
`

On the contrary, yaml file does not seem to work. I don't remember the error message, but it was about the fact that groups were not defined.

marcor4, 2019-02-20:
This was my configuration yaml:

```python
connection:
    tunneling:
         host: '192.168.1.110'
         port: 3671
         local_ip: '192.168.1.103'
```

Now I am noticing that https://xknx.io/configuration uses `xknx:` instead of `connection:`. Is it making any difference?

farmio, 2019-02-20:
https://xknx.io/home_assistant is using `xknx:` this is the Home-Assistant plugin. If you use xknx directly or pass a yaml file to xknx you will need `connection:`

If you get the current master from github (or wait til 0.9.5 is released) this configuration yaml should almost work. Almost because `host:` is the HA-plugin-term - for xknx this would be `gateway_ip:`; `port:` would be `gateway_port:`

marcor4, 2019-02-20:
Yes correct. I posted the wrong yaml file (one of my attempts with the HA, which I am not using anymore). So with the newest version the xknx the following yaml shoud work:

```python
connection:
     tunneling:
          gateway_ip: '192.168.1.110'
          gateway_port: 3671
          local_ip: '192.168.1.103'
```

farmio, 2019-02-20:
Exactly.

closed 2019-02-20 by farmio