XKNX/xknx #133: On Windows : asyncio\events.py", line 481, in add_signal_handler raise NotImplementedError

pulse-mind, 2018-08-23:
Hello,

When I run an example (example_telegram_monitor.py) under windows I get this error. 

Thank you
Fred

pulse-mind, 2018-08-24:
Thank you for your answer. 
How I do that ? "did you try to connect directly via a configured tunnel instead of using the gateway scanner"

Julius2342, 2018-08-24:
try to pass a connection config where you have set the mode to "Tunnel" and have speficied a fixed IP for the gateway.

https://github.com/XKNX/xknx/blob/master/xknx/xknx.py#L63

pulse-mind, 2018-08-24:
Thank you. I was already using the ConnectionConfig for setting IP and Port. I changed AUTOMATIC to TUNNELING. The error is still the here
```
asyncio\events.py", line 481, in add_signal_handler
    raise NotImplementedError
NotImplementedError
Task was destroyed but it is pending!
```

Julius2342, 2018-08-24:
uh .. what python version do you use?

The error is when registering the sig-int handler (which helps cleaning up after having received a control-C signal). May I ask you to checkout xknx, use it with PYTHON_PATH and comment out this line?

pulse-mind, 2018-08-24:
I am using Python 3.6

Julius2342, 2018-08-24:
https://github.com/crossbario/autobahn-python/commit/231c204eda00b139d259073bcfe864bf8393fc2f

looks like windows does not support signals ...

May you try not using daemon_mode (`daemon_mode=False`) and instead simply add an endless loop with an async-sleep?

Julius2342, 2018-08-24:
We also could do something like:

```
try:
    loop.add_signal_handler(signal.SIGTERM, loop.stop)
except NotImplementedError:
    pass  # Ignore if not implemented. Means this program is running in windows.
```

pulse-mind, 2018-08-24:
This is what I done locally

```
from sys import platform
```

```
if platform == "win32":
            self.logger.warning('Windows does not support signals')
        else:
            self.loop.add_signal_handler(signal.SIGINT, sigint_handler)
```

Then it works fine

pulse-mind, 2018-08-24:
Do you want me to commit something ?

Julius2342, 2018-08-24:
Yay! A pull request is highly appreciated!

pulse-mind, 2018-08-24:
I did a pull request. 

Also I have a question. 
I'd like to monitor the bus and store some (for some address group) of the values in database. 
I started with the example_telegram_monitor.py. I've seen the method telegram_received_cb where I get a telegram. A telegram has a payload. In some case the payload is a DPTArray. 
1/ how do I get the real value ? like the temperature of the sensor ? 
2/ is that possible to use the devices define in xknx.yaml, watch for telegram or something else (I've seen the def register_device_updated_cb) and then get the value of the device ? 

Thank you
Fred

Julius2342, 2018-08-24:
I would instanciate a `Sensor` device and listen to changes via callback inside this device.

closed 2020-01-21 by farmio