XKNX/xknx #134: Problem with :expose and update done once every hour

markuska00, 2018-08-23:
I use knx: expose to expose a variable from the component "hass variables". 
The variable is in the type "percent" and is used to control sun curtains.

It works well besides once every hour, the knx component sends some kind of update to the KNX bus.
The problem is that the knx component first sends an empty telegram that is perceived as 0%. 
Then after 0.1 seconds later send the correct telegram with the variable.

For me the sun blind closes first and then goes back to the correct postition.

Is this a bug?

Can I disable the "update" done every hour?

Julius2342, 2018-08-23:
yes,there is a parameter: - https://github.com/XKNX/xknx/blob/master/xknx/xknx.py#L61

If you use HASS there is a config option for this.

markuska00, 2018-08-25:
Anyone else getting the same problem in HASS if you disable state_updater?

markuska00, 2018-09-03:
Is this a bug that will be fixed?

Or is there any other way I can disable the update that happens once per hour without disabling all status updating?

Julius2342, 2018-09-03:
@markuska00 : may you please try out if #140 solves your problem?

markuska00, 2018-09-04:
@Julius2342: I can try  :-)
But does :expose also have state feedback?

markuska00, 2018-09-11:
#140 did not solve the problem with :expose
There is no changes in the "expose_sensor.py"

closed 2020-01-21 by farmio